indilib/indi #1500: INDI 1.9.1 throwing GPF faults

sonny486, 2021-06-27:
indi_asi_ccd[4241] general protection fault ip:7fa44f473870 sp:7ffd17842d10 error:0 in libc-2.31.so[7fa44f3fb000+178000]

indi_qhy_ccd[1872]: segfault at fffffff9 ip 00007fa361f9e870 sp 00007fff2480cb20 error 4 in libc-2.31.so[7fa361f26000+178000]Jun 27 14:58:13 astro-comp kernel: [   42.036295] Code: ff 49 89 ec e9 76 fd ff ff 90 f3 0f 1e fa 48 83 ec 18 48 8b 05 99 d6 14 00 48 8b 00 48 85 c0 0f 85 7d 00 00 00 48 85 ff 74 70 <48> 8b 47 f8 48 8d 77 f0 a8 02 75 34 48 8b15 f5 d4 14 00 64 48 83

knro, 2021-06-27:
This happens when? when starting the driver? or connecting to camera? I cannot reproduce it here.

sonny486, 2021-06-27:
For me when starting 1.9.1, HOWEVER I could not update my K-Stars to the latest nightly due to loop error.

Will try again tonight here in the US.

sonny486, 2021-06-27:
Multiple users reporting the issue too.

https://indilib.org/forum/development/9896-getting-general-protection-faults-in-all-the-drivers-i-load-1-9-1-update-libc-2-31-so.html

closed 2021-06-27 by sonny486