indilib/indi #1502: AZGTi mount control always moves on the same direction

miceno, 2021-06-28:
**Describe the bug**
Using `indi_azgti_telescope`, when I start Ekos to control the mount, open the telescope control to move the mount, but whatever the arrow button I click, it always moves in the same direction.

If at the same time I connect to the mount using SynScanPro (by Wifi, not using the Ekos software), the mount moves properly.

**To Reproduce**
Exact steps to reproduce the behavior.
1.  Open Ekos, and open the profile that includes the AZGTi driver `indi_azgti_telescope`
2. Connect via Wifi to the mount. I am using AZGTi station mode, and the mount connects to an [astroberry](https://www.astroberry.io/) hotspot.
2. Open the Mount Control Panel.
3. Move the mount by clicking using whatever direction arrow.
4. Same behaviour if you move the mount from the mount panel on INDI control panel.

**Expected behavior**
The mount should move according to the arrow clicked on the Mount Control Panel or the direction clicked on the INDI control panel.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: astroberry (latest 2.0.3).
 - Kstars: 3.5.3, but also happened in 3.5.2

**Log Files**
Pending to reproduce with logs.

knro, 2021-07-24:
I tested yesterday and I could not reproduce this behavior.

miceno, 2021-07-26:
Sorry for not providing more details, but it is hard... I am using it without any problem, and then suddenly, the arrows on the mount control only work on one direction, it is usually just one direction that is not working, not both directions at the same time.

It happened again tonight, so I had to use the SynScan Pro app instead of Ekos/Kstars to move the telescope.

One of those times, I managed to solved it, but I am not sure what I did 😞 , I think it was related to syncing or activating something on the tracking, but I don't know exactly what I did. 

I will try to get logs in the next session, but it was a little bit frustrating, I only have 1-2 hours in the evening to play with it, I will try to test it during the day.

miceno, 2021-07-26:
The issue is also in Kstars 3.5.4.

miceno, 2021-08-06:
Closing as I don't have logs. In case I will get the logs, I will reopen it again.

closed 2021-08-06 by miceno