indilib/indi #1504: INDI fails to build on Launchpad for 21.04+

knro, 2021-06-29:
Been getting failures for x86-64 and arm64 when compiling INDI on [Launchpad](https://code.launchpad.net/~mutlaqja/+recipe/libindi-nightly-unstable).

This happens in the linking step and they are all related to relocation error. That's what I got in a VM:

```
[  0%] Linking CXX shared library libindidriver.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/indidriver.dir/link.txt --verbose=1
/usr/bin/g++ -fPIC -g -O2 -ffile-prefix-map=/home/jasem/packages/indi=. -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -Wl,-Bsymbolic-functions -flto=auto -Wl,-z,relro  -Wl,-z,nodump -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -pie  -shared -Wl,-soname,libindidriver.so.1 -o libindidriver.so.1.9.1 CMakeFiles/indidriver.dir/indidriver.c.o CMakeFiles/indidriver.dir/libs/webcam/v4l2_decode/v4l2_builtin_decoder.cpp.o CMakeFiles/indidriver.dir/libs/indibase/hid_libusb.c.o  -lusb-1.0 -lnova -lpthread -lcfitsio -lm -lz -ljpeg -lfftw3 -ltheoraenc -ltheoradec -logg 
/usr/bin/ld: /tmp/libindidriver.so.1.9.1.MKiPkN.ltrans6.ltrans.o: warning: relocation against `_ZTINSt8__detail11_AnyMatcherINSt7__cxx1112regex_traitsIcEELb1ELb0ELb0EEE' in read-only section `.text'
/usr/bin/ld: /tmp/libindidriver.so.1.9.1.MKiPkN.ltrans0.ltrans.o: relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
```

[log.txt](https://github.com/indilib/indi/files/6731712/log.txt)

However, we are already setting **set(CMAKE_POSITION_INDEPENDENT_CODE ON)** in line 9 in CMakelists.txt, so any ideas what's causing this error for newer version of GCC apparently?

@jpaana @jochym @pawel-soja @xsnrg  any ideas?

k-ross, 2021-06-29:
Could it be because of PIE flags? The documentation states "These options are similar to -fpic and -fPIC, but generated position independent code can be only linked into executables."

jpaana, 2021-06-29:
I compile with 21.04 on x64 and default compiler daily without problems as my dev and observatory machines run that, but haven't seen this one. Differences to my compile options seem to be lto and hardening related. 21.04 is the first version with lto turned on by default for packages so I tried adding -flto=auto to my build options and got the same error, so that seems to be the culprit (or symptom of some other problem).

knro, 2021-06-29:
> Could it be because of PIE flags? The documentation states "These options are similar to -fpic and -fPIC, but generated position independent code can be only linked into executables."

That's **exactly** it, it now makes a valid debian package.

@jpaana compiling works OK, it was failing in creating a debian package via *debuild -uc -nc*

closed 2021-06-29 by knro

jpaana, 2021-06-29:
Good, I just got the same error while linking from my normal build with just adding the -flto=auto, but yes, your latest commit fixed that too.

commit 7fb682d3 (2021-06-29): Move CMAKE_POSITION_INDEPENDENT_CODE to CMakeCommon and disable -pie. Fixes #1504