indilib/indi #1517: indi-3rdparty/indi-gphoto/gphoto_driver.cpp incompatibility with LibRaw 0.20.2

jmocenigo, 2021-07-24:
LibRaw 0.20.2 changed the location of CameraTemperature and SensorTemperature from imgdata.other to imgdata.makernotes.common. The four references to "other" in gphoto_driver.cpp need to be changed (or conditionally handled via changes to LIBRAW_CAMERA/SENSOR_TEMPERATURE defines) to "makernotes.common" to allow compilation with LibRaw 0.20.2 and later

knro, 2021-07-24:
Thanks, can you please submit a PR to fix this?

jmocenigo, 2021-07-24:
OK, I am starting on that now, but I can only make the “dumb” code change and not a more robust conditional compilation change (based on available LibRaw version) as I am not sufficiently familiar with autoconf to pull that off properly.

jmocenigo, 2021-07-24:
Apologies.

My source tree must not have been up-to-date.

When I went through the process of filing a PR, I used an entirely new cloned tree and it compiled properly. I also saw that in the appropriate CMakeLists.txt there already is code that checks both the “other” and “common” LibRaw structures when building conditional compilation variables.

Sorry for the false alarm. The problem showed up right after I updated to Ubuntu 21.04 so I thought it was related to that somehow.

Next time I will follow the PR process (https://indilib.org/develop/tutorials/181-how-to-contribute-to-indi-github-development.html <https://indilib.org/develop/tutorials/181-how-to-contribute-to-indi-github-development.html>) before filing an issue.

closed 2021-07-24 by knro