indilib/indi #1519: Synscan coordinate errors (Skywatcher Dobsonian altaz mount)

ajeb78, 2021-07-25:
**Describe the bug**
Affects indi_synscan. I am using indi_synscan with a Skywatcher 250P Dobsonian on a GOTO altaz mount. The mount itself behaves correctly when used only with the handset and once aligned will slew to a given GOTO object reasonably accurately.
However when aligned and then controlled via the indi_synscan module it slews to incorrect co-ordinates. It appears to be slewing close to the correct declination but about 3 hours out in RA, (not sure if this error is always the same or just with the objects I tested on). The time on both my laptop and mount is set in UTC with DST turned off, but the bug also occurs if all times are set to BST with DST turned on. My lat / long is approx 54N 003W (this is manually input correctly into the handset from GPS coordinates from my phone).

**To Reproduce**
Align telescope using handset.
GOTO known object.
Connect telescope to computer using USB cable provided.
Run driver (via kstars / ekos).
GOTO other object via kstars.
Telescope position on kstars does not agree with position on handset, telescope points incorrectly.

**Expected behavior**
Telescope position should agree between kstars and telescope hand control.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**

    OS: Kubuntu 21.04
    INDI Library: 1.9.1
    Code 1.9.1-tgz. Protocol 1.7.
    Synscan handset version 4.39.15

knro, 2021-07-25:
Can please you please turn on INDI + Mount logging and attach the logs?

ajeb78, 2021-07-26:
Hopefully this provides more information. It's not dark at the moment here so I've just aligned the telescope to the approximate direction of Polaris and accepted the alignment when it slews to Arcturus, but the behaviour appears consistent.

[log_17-57-06.txt](https://github.com/indilib/indi/files/6880311/log_17-57-06.txt)

I've also attached some screenshots showing the setup pages in case it helps (or if it lets you spot that I'm doing something dumb!)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

As before, the displayed location of the mount in kstars jumps all over the place as the telescope is moving and then comes to rest in an incorrect position.

If you need any more information just let me know.

knro, 2021-07-27:
It's fixed, sorry for the stupid bug. Try the nightly/beta or compile from GIT if you know how.

closed 2021-07-27 by knro

ajeb78, 2021-07-27:
Fix confirmed, thank you :)

commit 96fc2928 (2021-07-27): Convert RA from Degrees --> Hours. Fixes #1519