indilib/indi #1528: libindiclient leaks last blob on driver restart

pludov, 2021-08-29:
Upon removal of devices or property, blob data are not freed. As a result, a memory leak of blob will occurs when a driver that has received blob is restarted:

1. Compile the dummy-streamer example program joined
2. Start an indi server  with ccd_simulator
3. Start dummy-streamer
4. Activate streaming on the ccd_simulator: dummy-streamer logs it blob receptions
5. Restart the ccd_simulator (through indi fifo)
6. Restart streaming on the ccd_simulator

Repeating 5&6 will make dummy-streamer program memory size. You can use bigger frame size to make the problem more obvious. Alternatively, running dummy-streamer with valgrind will report the following leak:

``` 
==26462== 1,328,925 bytes in 1 blocks are definitely lost in loss record 28 of 28
==26462==    at 0x4C33D2F: realloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==26462==    by 0x11FCC3: INDI::BaseDevice::setBLOB(_IBLOBVectorProperty*, xml_ele_*, char*) (basedevice.cpp:731)
==26462==    by 0x122837: INDI::BaseDevice::setValue(xml_ele_*, char*) (basedevice.cpp:692)
==26462==    by 0x10FF1D: INDI::BaseClientPrivate::dispatchCommand(xml_ele_*, char*) (baseclient.cpp:579)
==26462==    by 0x110CF5: INDI::BaseClientPrivate::listenINDI() (baseclient.cpp:440)
==26462==    by 0x53376DE: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25)
==26462==    by 0x4E456DA: start_thread (pthread_create.c:463)
==26462==    by 0x5CDA71E: clone (clone.S:95)
```

My though: blob data freeing is commented out in indipropertyview.h, so there is simply no free on removal. I am not sure the impact of activating it back however.

```
template <>
struct WidgetView<IBLOB>: PROPERTYVIEW_BASE_ACCESS IBLOB
{
    ...
    ~WidgetView()                                          { /* free(this->blob); */ }
    void clear()                                           { /* free(this->blob); */ memset(this, 0, sizeof(*this)); }
```

knro, 2021-08-29:
It should be clear. Not sure what @pawel-soja has intended exactly with this? Can you uncomment and test it out? Maybe because the BLOB could be shared with the client still so freeing the buffer would cause issues in the client at this stage... but we are responsible for freeing any memory we allocate.

knro, 2021-09-28:
Ok looks like @pawel-soja is busy. Did you try uncommenting the free and check for any possible regressions?

knro, 2022-11-22:
@pawel-soja was this issue fixed in your latest PR?

pawel-soja, 2022-11-22:
@knro sure i will check it out, but after i finish rework in CMakeLists.

knro, 2023-02-27:
@pawel-soja Any confirmation on whether this issue is resolved?

pawel-soja, 2023-02-27:
Hey, I analyzed the problem.

The `WidgetView` class cannot free memory because it is just an access frontend. When creating these classes, I left the behavior as it was before, i.e. whoever created the Blob managed the memory for the data. My goal was to create `INDI::PropertyXXX` that manages the allocation completely, but in the case of Blob there were some difficulties, each piece of code allocates and frees memory differently, additionally there was an additional complexity in the form of:
https://github.com/indilib/indi/blob/fe39e52d30e2ec37cb5ffd94075a529beb769211/libs/indidevice/basedevice.cpp#L796-L800

`indi/libs/indibase/stream/encoder/rawencoder.cpp`
[image]

So, as I mentioned, I left the Property behavior unchanged so as not to free someone's memory.

There are 2 choices, either fix the `BaseClient` implementation so that it frees memory when the Property is destroyed, or, more future-proof, create a memory management implementation on the Property side. Unfortunately, with the second option, I can't guarantee that something won't crash with foreign programs that allocate memory and give a pointer to PropertyBlob.

pawel-soja, 2023-02-27:
There is also an option to add the `setBlobDeleter` function so that the programmer decides how the memory should be freed.
```cpp
// A
{
    INDI::PropertyBlob blob{1};
    blob.setBlobDeleter([](void *data)
    {
        free(data); // correct for malloc
    });

    blob[0].setBlob(malloc(128));
}

// B
{
    INDI::PropertyBlob blob{1};
    blob.setBlobDeleter([](void *data)
    {
        delete[] reinterpret_cast<char*>(data); // correct for new operator[]
    });

    blob[0].setBlob(new char[128]);
}
```

knro, 2023-03-01:
I like this deleter option as it gives more freedom on how should be cleared on property destruction.

closed 2024-04-19 by github-actions[bot]