indilib/indi #1531: EKOS crashes when connecting to SX INDI drivers

Brnikat, 2021-09-07:
I can't find anything about this one on the net. If it is a known problem, please point me at the solution. If more information would be useful, please let me know and I will try to find out.  Thanks in advance.  Paul

The telescope mount is controlled by a FS-2 which emulates a Meade LX200. he imaging train is a SX Trius 814 behind a SX MAXI USB filterwheel. Guiding is (supposed to be) by a SX Lodestar 2 attached to a separate guide scope.

The TCS is a small Odroid box running Ubuntu 20.04 LTS. All packages are up to date.  These outputs may prove informative:

odroid@odroid:~$ uname -a
Linux odroid 4.9.277-117 #1 SMP PREEMPT Sat Aug 7 21:43:06 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

odroid@odroid:~$ lsusb
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04d9:1603 Holtek Semiconductor, Inc. Keyboard
Bus 001 Device 009: ID 1278:0920 Starlight Xpress 
Bus 001 Device 008: ID 1278:0507 Starlight Xpress Lodestar autoguider
Bus 001 Device 007: ID 1278:0198 Starlight Xpress USB2.1 Hub
Bus 001 Device 005: ID 04b4:6572 Cypress Semiconductor Corp. Unprogrammed CY7C65642 hub
Bus 001 Device 004: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 003: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The following output from gdb followed by a backtrace gives this when EKOS is fired up and the INDI server started.

odroid@odroid:~$ gdb -ex run --args kstars
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kstars...
(No debugging symbols found in kstars)
Starting program: /usr/bin/kstars 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fafaeee30 (LWP 6819)]
org.kde.kstars: Welcome to KStars 3.4.1
org.kde.kstars: Build: 2020-02-24T06:23:39Z
org.kde.kstars: OS: "ubuntu"
org.kde.kstars: API: "arm64-little_endian-lp64"
org.kde.kstars: Arch: "arm64"
org.kde.kstars: Kernel Type: "linux"
org.kde.kstars: Kernel Version: "4.9.277-117"
org.kde.kstars: Qt Version: 5.12.5
[New Thread 0x7faf2a1e30 (LWP 6820)]
[New Thread 0x7facf2be30 (LWP 6821)]
[New Thread 0x7f9af73e30 (LWP 6822)]
[New Thread 0x7f93ffee30 (LWP 6823)]
[New Thread 0x7f9a772e30 (LWP 6824)]
[New Thread 0x7f99f71e30 (LWP 6825)]
[New Thread 0x7f99770e30 (LWP 6826)]
[New Thread 0x7f98f6fe30 (LWP 6827)]
[New Thread 0x7f937fde30 (LWP 6828)]
[New Thread 0x7f92ffce30 (LWP 6829)]
[New Thread 0x7f927fbe30 (LWP 6830)]
[New Thread 0x7f91ffae30 (LWP 6831)]
[New Thread 0x7f917f9e30 (LWP 6832)]
[New Thread 0x7f90ff8e30 (LWP 6833)]
[New Thread 0x7f63ffee30 (LWP 6834)]
[New Thread 0x7f637fde30 (LWP 6835)]
[... 334 lines cut ...]
#4  0x00000055558c5394 in  ()
#5  0x00000055558ca164 in  ()
#6  0x000000555586af04 in  ()
#7  0x0000007fb59d85b4 in QObject::event(QEvent*) () at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#8  0x0000007fb64ac3fc in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
    at /usr/lib/aarch64-linux-gnu/libQt5Widgets.so.5
#9  0x0000007fb64b5980 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/aarch64-linux-gnu/libQt5Widgets.so.5
#10 0x0000007fb59aaeb0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
    at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#11 0x0000007fb59ade00 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
    at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#12 0x0000007fb5a06a40 in  () at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#13 0x0000007fb455f944 in g_main_context_dispatch () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0
#14 0x0000007fb455fbb4 in  () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0
#15 0x0000007fb455fc54 in g_main_context_iteration () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0
#16 0x0000007fb5a05ef4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#17 0x0000007fb59a9698 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#18 0x0000007fb59b1cac in QCoreApplication::exec() () at /usr/lib/aarch64-linux-gnu/libQt5Core.so.5
#19 0x0000005555634918 in  ()
#20 0x0000007fb4b7a090 in __libc_start_main (main=
    0x5555632fd8, argc=1, argv=0x7ffffff3f8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:308
#21 0x000000555565c880 in  ()
(gdb)

knro, 2021-09-08:
Thanks for the GDB log but unfortunately it does not shed light on the issue. I suggest you use up to date KStars (or any other client) as the version you are using is a bit old (3.4.1 vs current 3.5.4)

Brnikat, 2021-09-08:
Thanks.  Performed full upgrade to Ubuntu 21-04 and  the curent version should be installed. Unfortunately there are "held broken packages" preventing the install.

root@odroid:/home/odroid# apt --fix-broken install kstars
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libkf5configwidgets5 : Depends: libkf5guiaddons5 (>= 5.80.0~) but it is not installable
 libkf5xmlgui5 : Depends: libkf5guiaddons5 (>= 5.80.0~) but it is not installable
                 Depends: qtbase-abi-5-15-2
                 Recommends: libkf5xmlgui-bin (= 5.80.0-0ubuntu1) but it is not going to be installed
 libqt5printsupport5 : Depends: qtbase-abi-5-15-2
 libqt5qml5 : Depends: qtbase-abi-5-15-2
 libqt5qmlmodels5 : Depends: qtbase-abi-5-15-2
 libqt5quick5 : Depends: qtdeclarative-abi-5-12-8
 libqt5sql5 : Depends: qtbase-abi-5-15-2
 libqt5sql5-sqlite : Depends: qtbase-abi-5-15-2
 libqt5svg5 : Depends: qtbase-abi-5-15-2
 libqt5texttospeech5 : Depends: qtbase-abi-5-15-2
                       Recommends: qtspeech5-speechd-plugin but it is not going to be installed or
                                   qtspeech5-flite-plugin but it is not going to be installed
 libqt5websockets5 : Depends: qtbase-abi-5-15-2
 libqt5xml5 : Depends: qtbase-abi-5-15-2
 qml-module-qtquick-controls : Depends: qml-module-qtgraphicaleffects but it is not going to be installed
                               Depends: qml-module-qtquick-window2 but it is not going to be installed
                               Depends: libqt5quick5 (>= 5.15.2+dfsg~) but 5.12.8-0ubuntu1 is to be installed or
                                        libqt5quick5-gles (>= 5.15.2+dfsg~) but it is not installable
                               Depends: qtbase-abi-5-15-2
 qml-module-qtquick-layouts : Depends: qtbase-abi-5-15-2
 qml-module-qtquick2 : Depends: libqt5quick5 (>= 5.15.1) but 5.12.8-0ubuntu1 is to be installed or
                                libqt5quick5-gles (>= 5.15.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
root@odroid:/home/odroid# 

Oh well, now trying to fix that problem.

Brnikat, 2021-09-08:
Installed kstars-bleeding (a lengthy ordeal in its own right) with exactly the same crashing behaviour.

odroid@odroid:~$ kstars
org.kde.kstars: Welcome to KStars 3.5.4 Stable
org.kde.kstars: Build: 2021-07-08T11:31:02Z
org.kde.kstars: OS: "ubuntu"
org.kde.kstars: API: "arm64-little_endian-lp64"
org.kde.kstars: Arch: "arm64"
org.kde.kstars: Kernel Type: "linux"
org.kde.kstars: Kernel Version: "4.9.277-117"
org.kde.kstars: Qt Version: 5.15.2
...

The SX814 and Lodestar2 appear in a window briefly when the INDI drivers are started but disappear when the driver crashes.

Not sure what to try next but I am very willing to give reasonable suggestions a try.  What is so annoying is that it used to work under 18.04 LTS, which makes me wonder if the USB behaviour changed between 18.04 and 20.04 (and now 21.04).

Downgrading to 18.04 would be a real PITA but I may be able to press a AMD-64 Linux box into temporary service to see whether it is just the ARM64 environment which may be broken.

knro, 2021-09-08:
Does the driver itself crash? Use this script:

```
#!/bin/bash

filename=~/crash_$(date +%Y%m%d_%H%M%S).txt
echo "Starting INDI driver debugging for" $*
gdb -batch -ex "set follow-fork-mode child" -ex "run" -ex "bt" --args indiserver -r 0 -v $* > $filename 2>&1
echo "Debugging finished. Please send" $filename "file to support."

```

Save it as indigdb.sh (for example) then run it like this

./indigdb.sh indi_sx_wheel

if the driver crashes, it's supposed to save a crash file that you can post here.

Brnikat, 2021-09-09:
That didn't crash exactly, but terminated when C-Z was typed because it received a SIGSTP.
 The attempt with ./indigdb.sh indi_sx_wheel & immediately killed the shell!  After logging in again this was in crash_20210909_093112.txt

odroid@odroid:~$ cat crash_20210909_093112.txt
2021-09-09T09:31:12: startup: /usr/bin/indiserver -r 0 -v indi_sx_wheel 
[Attaching after process 27645 fork to child process 27648]
[New inferior 2 (process 27648)]
[Detaching after fork from parent process 27645]
2021-09-09T09:31:12: Driver indi_sx_wheel: pid=27648 rfd=3 wfd=6 efd=7
2021-09-09T09:31:12: listening to port 7624 on fd 4
[Inferior 1 (process 27645) detached]
process 27648 is executing new program: /usr/bin/indi_sx_wheel
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
odroid@odroid:~$ 

Not too sure why you asked for the wheel driver.  It is the SX814 and Lodestar2 devices which appear only briefly in EKOS as noted above. The FW remains behind.  Sorry if this latter wasn't clear in my earlier material.

Brnikat, 2021-09-09:
This what happens with ./indigdb.sh indi_sx_ccd

2021-09-09T09:43:04: startup: /usr/bin/indiserver -r 0 -v indi_sx_ccd 
[Attaching after process 27773 fork to child process 27776]
[New inferior 2 (process 27776)]
[Detaching after fork from parent process 27773]
2021-09-09T09:43:04: Driver indi_sx_ccd: pid=27776 rfd=3 wfd=6 efd=7
2021-09-09T09:43:04: bind: Address already in use
2021-09-09T09:43:04: good bye
[Inferior 1 (process 27773) detached]
process 27776 is executing new program: /usr/bin/indi_sx_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb58cdd10 (LWP 27777)]

Thread 2.2 "libusb_event" received signal SIG33, Real-time event 33.
[Switching to Thread 0x7fb58cdd10 (LWP 27777)]
0x0000007fb7ae0de4 in __GI___poll (fds=0x7fb58cd448, nfds=2, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:41
41	../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
#0  0x0000007fb7ae0de4 in __GI___poll (fds=0x7fb58cd448, nfds=2, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:41
#1  0x0000007fb7ddbdbc in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#2  0x0000007fb765be84 in start_thread (arg=0x0) at pthread_create.c:473
#3  0x0000007fb7aead1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:77

knro, 2021-09-09:
Modify script from:

`-ex "set follow-fork-mode child"`

To:

`-ex "set follow-fork-mode child" -ex "handle SIG33 nostop noprint"`

then run again, it's safe to ignore SIG33

Brnikat, 2021-09-09:
After ensuring that no indi processes were still running, ./indigdb.sh indi_sx_ccd produced this crash file essentially immediately, suggesting to me that the driver did indeed crash:

odroid@odroid:~$ cat /home/odroid/crash_20210909_094634.txt
2021-09-09T09:46:34: startup: /usr/bin/indiserver -r 0 -v indi_sx_ccd 
[Attaching after process 27797 fork to child process 27800]
[New inferior 2 (process 27800)]
[Detaching after fork from parent process 27797]
2021-09-09T09:46:34: Driver indi_sx_ccd: pid=27800 rfd=3 wfd=6 efd=7
2021-09-09T09:46:34: listening to port 7624 on fd 4
[Inferior 1 (process 27797) detached]
process 27800 is executing new program: /usr/bin/indi_sx_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb58cdd10 (LWP 27801)]

Thread 2.2 "libusb_event" received signal SIG33, Real-time event 33.
[Switching to Thread 0x7fb58cdd10 (LWP 27801)]
0x0000007fb7ae0de4 in __GI___poll (fds=0x7fb58cd448, nfds=2, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:41
41	../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
#0  0x0000007fb7ae0de4 in __GI___poll (fds=0x7fb58cd448, nfds=2, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:41
#1  0x0000007fb7ddbdbc in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#2  0x0000007fb765be84 in start_thread (arg=0x0) at pthread_create.c:473
#3  0x0000007fb7aead1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:77
2021-09-09T09:46:35: Driver indi_sx_ccd: stderr EOF
2021-09-09T09:46:35: Driver indi_sx_ccd: Terminated after #0 restarts.
2021-09-09T09:46:35: good bye

Brnikat, 2021-09-10:
Hmm, this doesn't appear to have got throught the email interface, unlike past experience.  Let's try again...

This time was as for indi_sx_wheel earlier  That is, it didn't terminate
until sent a STOP signal and it killed the ssh session when run in the
background.

The processes from the second try are still running:
8<--------------------------------------------------------------------->8
droid@odroid:~$ ps axlww | grep indi
0   118    2994    2858  20   0 791784  7088 -      Ssl  ?          0:00
/usr/libexec/ayatana-indicator-session/ayatana-indicator-session-service
0   118    2996    2858  20   0 364064  8844 -      Ssl  ?          0:01
/usr/libexec/ayatana-indicator-datetime/ayatana-indicator-datetime-service
0   118    3009    2858  20   0 455160  4268 -      Ssl  ?          0:00
/usr/libexec/ayatana-indicator-power/ayatana-indicator-power-service
0   118    3014    2858  20   0 843964  6220 -      Ssl  ?          0:00
/usr/libexec/ayatana-indicator-sound/ayatana-indicator-sound-service
0   118    3018    2858  20   0 256704  4996 -      Ssl  ?          0:00
/usr/libexec/ayatana-indicator-application/ayatana-indicator-application-service
0   118    3024       1  20   0 471360 18284 -      Sl   ?          0:00
nm-applet --indicator
0  1000   27863       1  20   0   5620  3040 do_wai S    ?          0:00
/bin/bash ./indigdb.sh indi_sx_ccd
0  1000   27865   27863  20   0 569360 75928 poll_s Sl   ?          0:01
gdb -batch -ex set follow-fork-mode child -ex handle SIG33 nostop
noprint -ex run -ex bt --args indiserver -r 0 -v indi_sx_ccd
0  1000   27873   27865  20   0   2068   612 poll_s S    ?          0:00
/usr/bin/indiserver -r 0 -v indi_sx_ccd
0  1000   27876   27873  20   0  48940 11356 poll_s Sl   ?          0:00
indi_sx_ccd
0  1000   27982   27968  20   0   4992   804 pipe_w S+   pts/1      0:00
grep --color=auto indi
odroid@odroid:~$
8<--------------------------------------------------------------------->8

The crash log from the first run is as follows:

8<--------------------------------------------------------------------->8
odroid@odroid:~$ cat crash_20210909_105826.txt
2021-09-09T10:58:27: startup: /usr/bin/indiserver -r 0 -v indi_sx_ccd
[Attaching after process 27857 fork to child process 27860]
[New inferior 2 (process 27860)]
[Detaching after fork from parent process 27857]
2021-09-09T10:58:27: Driver indi_sx_ccd: pid=27860 rfd=3 wfd=6 efd=7
2021-09-09T10:58:27: listening to port 7624 on fd 4
[Inferior 1 (process 27857) detached]
process 27860 is executing new program: /usr/bin/indi_sx_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb58cdd10 (LWP 27861)]
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Telescope
Simulator.EQUATORIAL_EOD_COORD
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Telescope
Simulator.TELESCOPE_INFO
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Telescope
Simulator.GEOGRAPHIC_COORD
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Telescope
Simulator.TELESCOPE_PIER_SIDE
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Rotator
Simulator.ABS_ROTATOR_ANGLE
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Focuser
Simulator.ABS_FOCUS_POSITION
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on Focuser
Simulator.FOCUS_TEMPERATURE
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on CCD
Simulator.FILTER_SLOT
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on CCD
Simulator.FILTER_NAME
2021-09-09T10:58:27: Driver indi_sx_ccd: snooping on SQM.SKY_QUALITY

Thread 2.1 "indi_sx_ccd" received signal SIGTSTP, Stopped (user).
[Switching to Thread 0x7fb58d1620 (LWP 27860)]
0x0000007fb7ae39ec in __GI___select (nfds=<optimized out>,
readfds=0x7ffffff0c8, writefds=0x0, exceptfds=0x0, timeout=<optimized
out>) at ../sysdeps/unix/sysv/linux/select.c:49
49	../sysdeps/unix/sysv/linux/select.c: No such file or directory.
#0  0x0000007fb7ae39ec in __GI___select (nfds=<optimized out>,
readfds=0x7ffffff0c8, writefds=0x0, exceptfds=0x0, timeout=<optimized
out>) at ../sysdeps/unix/sysv/linux/select.c:49
#1  0x0000007fb7e41230 in ?? () from
/lib/aarch64-linux-gnu/libindidriver.so.1
#2  0x0000007fb7e429c4 in eventLoop () from
/lib/aarch64-linux-gnu/libindidriver.so.1
#3  0x0000007fb7e432c8 in main () from
/lib/aarch64-linux-gnu/libindidriver.so.1
#4  0x0000007fb7a389d4 in __libc_start_main (main=0x7fb7e431a4 <main>,
argc=1, argv=0x7ffffff2e8, init=<optimized out>, fini=<optimized out>,
rtld_fini=<optimized out>, stack_end=<optimized out>) at
../csu/libc-start.c:332
#5  0x0000005555559c38 in _start ()

knro, 2021-11-10:
I don't see crash above, it was stopped by you (SIGSTOP) I presume this is what it means.

closed 2021-11-24 by knro