indilib/indi #1535: libplayerone not built for ubuntu 21.04

atoy40, 2021-09-16:
Into the indinighly PPA, the libplayerone package is not built for ubuntu 21.04 (so indi-playone not built too). This is needed to install iindi-full-1.9.2 meta package

note : gsc is also absent in the indinightly repo for 21.04

Thanks
Anthony

knro, 2021-09-28:
This should be fixed, can you double checked.

atoy40, 2021-09-29:
I can confirm indi-full 1.9.2 is now installed and it install libindi 1.9.3 which is available on the repo.

```
ii  libindi-data                                  1.9.3~202109270845~ubuntu21.04.1                                     all          Instrument-Neutral Device Interface library -- shared data
ii  libindi-dev                                   1.9.3~202109270845~ubuntu21.04.1                                     amd64        Instrument-Neutral Device Interface library -- development files
ii  libindi1:amd64                                1.9.3~202109270845~ubuntu21.04.1                                     amd64        Instrument-Neutral Device Interface library -- shared library
ii  indi-full                                     1.9.2~202109262227~ubuntu21.04.1                                     amd64        Instrument-Neutral Device Interface library - Full INDI
ii  indi-playerone                                0.2~202109281122~ubuntu21.04.1                                       amd64        INDI Driver for Playerone based cameras
ii  libplayerone                                  0.1~202109281524~ubuntu21.04.1                                       amd64        Library for Player One Astronomy's Cameras
```
Anthony

closed 2021-09-29 by atoy40