indilib/indi #1553: Support for Celestron CGX/CGX-L PC port

KokaneeYYZ, 2021-11-04:
**Is your feature request related to a problem? Please describe.**
I can't get INDI to connect to my Celestron CGX mount when it is connected directly to any of my Pi4B USB ports, or via a powered USB 3.0 hub.  The following error shows up in Kstars when I try to connect to the mount:

[ERROR] Serial read error: Timeout error
[ERROR] Failed to communicate with the mount, check the logs for details.

I've tried all baud rates available for the but still get timeout errors.  Driver name is the "Celestron GPS" (indi_celestron_gps V3.6).  Jasem at Stellarmate suggested that I reach out to the INDI developers to see whether the "... PC port on the CGX is compatible with Celestron "Aux" protocol.".

**Describe the solution you'd like**
Mount can be controlled via the PC port.

**Describe alternatives you've considered**
I connect to the CGX Hand Controller's USB mini port and can control the mount using the "Prolific USB to Serial Controller" driver.

**Additional context**
INDI Control Panel screenshot showing options being used, and the Timeout error:

[image]

gilesco, 2021-11-04:
I think the Celestron CGX driver only supports the USB handset on the Nextstar+

You could look at this driver: https://indilib.org/telescopes/celestron/celestron-aux-driver.html

This is a Beta driver only, still in testing, I believe they have said that the mount should be supervised while using this driver, in case of damaging behaviour.

knro, 2021-11-04:
I believe @KokaneeYYZ already tried the AUX driver without success also. Perhaps post for the logs for the AUX attempt?

knro, 2022-02-19:
This should be fixed against Celestron AUX driver in INDI 3rd party repo.

closed 2022-02-19 by knro