indilib/indi #1554: FocusLynx Syncing current issues and proposed improvement

DanielVanNoord, 2021-11-04:
With further testing on the FocusLynx driver I discovered odd behavior involving syncing. For many device types the driver forces the focuser to sync before it can be moved. Although this can be turned off (seemingly only for a single session do to an issue that I believe occurs in checkIfAbsoluteFocuser()) with the Sync Mandatory option I am unsure if this requirement it is needed. All Optec Lynx devices are absolute, they operate in steps and retain the position across power cycles and connections.

I suspect this issue arose over differences between Optec FocusLynx Homing and Syncing focusers. Homing focusers automatically move on powerup to calibrate the range and position. These focusers cannot sync. Syncing focusers allow the user to sync the position. However, once synced the focuser will retain the position unless the user manually disengages the focuser and manually moves it. At that point it is recommended that the user resync the focuser so the physical position matches the controllers position but this is considered optional.

Currently the “Absolute” check in checkIfAbsoluteFocuser() is checking for a mix of Homing and Syncing absolute focusers. This can be updated to be either Homing or Syncing with both being absolute and only Syncing supporting optional Sync commands.  

Proposed changes:
1.	Update list of homing vs syncing devices types to use the Device ID code for homing focusers. This can be read on connection to determine if the controller supports homing or syncing for the current focuser.
2.	Make syncing optional by default or remove the sync mandatory functions completely. As all focusers are absolute syncing should be optional for all that support it.
3.	Disable syncing for homing focusers. Currently allowing it should be harmless, the controller will return an error if these are synced and ignore the request. But is seems good to remove something that has no function.

I can work on developing and testing a pull request if you like the proposed changes. But I wanted some feedback to make sure that I am on the right track for how indi uses focusers. Do you have any thoughts?

knro, 2021-11-04:
Sounds good Daniel. I thought from reading the driver code that some focusers are purely relative (i.e. no absolute encoder or count), but it appears they're all absolute, but differ in homing vs. syncing? this applies to all the models supported by this driver?

DanielVanNoord, 2021-11-04:
Yep, all models that this driver supports maintain an internal count. The controllers maintain and track the position internally. The only difference is that some use automatic homing to find the zero point every time they turn on.

Is it OK if I work on and submit a patch for this?

knro, 2021-11-04:
Yes please go ahead. Another issue that a user report is that the model name is not saved properly or loaded on next driver restart? I checked the code and saw that it reads the model name from the controller configuration and then uses that to select which model it should be in the driver (the model dropdown). Is this the optimal way of doing this?

DanielVanNoord, 2021-11-04:
Yes, in fact I would always read the model from the controller and only allow it to be set when connected. Some controllers support a limited number of device types because they are integrated into the focuser that they support. In fact if I were to make two other proposals:
1. We can improve and possibly reduce the number of listed device types. Newer versions of Optec's drivers / firmware have reduced these down quite a bit as a fair number of devices use the same internal settings. We now use a generic device type for all DirectSync / Posi-Drive motors and all QuickSync / HSM motors. This was done in a backwards and forwards compatible way, but matching the current user documentation would be good.
2. Only allow the device type to be changed when connected and always use the type reported from the controller.

If you think this is OK I can also work on it as well. If so should I make it a separate issue report / pull request?

knro, 2021-11-04:
Great, let's do this in a separate PR!

beettlle, 2021-11-08:
Would it be possible to link the new PR to this one? I own a FocusLynx running with StellarMate so I’m very interested in this work.

PR #1558 (2021-11-08): Improve lynx sync and homing behavior
Requesting review for the following changes to the Lynx driver

1. Removes the required sync before move from all Lynx focusers as they are all absolute
2. Use the controllers returned device ID to determine support for Home vs Sync
3. Disable syncing for homing focusers

This should fix #1554 

Please let me know if there are any required or requested changes, I am learning the layout and style for INDI drivers.

So far I have tested this on a FocusLynx hub with a DirectSync and TCF-S 2 as well as a ThirdLynx and a TCF-LSI.
closed 2021-11-08 by knro