indilib/indi #1555: sestosenso driver

DazP666, 2021-11-06:
Any chance we can change the FocusABSPosN:Max from 200000 to 2000000? 
the focuser can go above 2million steps, not sure why someone limited it to 200,000?
anyone using a sestosenso with an SCT 10:1 microfocuser can't acheive focus, 
my focus position is around 1,100,000
the max number you can put in the focuser is 200,000.

I've been using Kstars for a few years now, never used to have this issue, only appeared since last year sometime.
My alternatives are dumping 2x expensive SestoSenso Focusers (so no thanks) or going back to SGP/NINA? another thing I'd rather not do, but Kstars is currently useless for my setup, I'd change software than gear.
Seems excessive for one "zero" but that's my predicament.

k-ross, 2021-11-06:
What should the maximum be? You said it can go over 2 million, but you're proposing the maximum should be 2 million.

k-ross, 2021-11-06:
Looking at the source, I see:

```FocusAbsPosN[0].min   = 0.;
    FocusAbsPosN[0].max   = 200000.;
    FocusAbsPosN[0].value = 0;
    FocusAbsPosN[0].step  = 1000;

    FocusMaxPosN[0].value = 2097152;
```

There's a read-only field setting the max to 2,097,152, but the AbsPos has a max of 200,000. I guess it should be the same as the max of 2,097,152.

DazP666, 2021-11-06:
FocusAbsPosN[0].min   = 0.;
    FocusAbsPosN[0].max   = 200000.;
    FocusAbsPosN[0].value = 0;
    FocusAbsPosN[0].step  = 1000;

    FocusMaxPosN[0].value = 2097152;

this is from the driver, the max focuser steps/position is 2,097,152
but the max I can set it to is 200,000 as in FocusAbsPosN.
my focus point is over a million, I used to be able to put that number in until last year

k-ross, 2021-11-06:
I'll create a PR to make that fix. Then when Jasem approves it, it should go into the next nightly build.

DazP666, 2021-11-06:
> Looking at the source, I see:
> 
> ```
>     FocusAbsPosN[0].max   = 200000.;
>     FocusAbsPosN[0].value = 0;
>     FocusAbsPosN[0].step  = 1000;
> 
>     FocusMaxPosN[0].value = 2097152;
> ```
> 
> There's a read-only field setting the max to 2,097,152, but the AbsPos has a max of 200,000. I guess it should be the same as the max of 2,097,152.

I figured that, not sure why 200,000 is in there?

DazP666, 2021-11-06:
> I'll create a PR to make that fix. Then when Jasem approves it, it should go into the next nightly build.

Thanks very much, It must only be me using a RASA11v1/EdgeHD11+SestoSenso+10:1 Microfocuser?
I had to remove the SestoSenso end of last season and manually focus instead, as its completely broken for me.. 
be nice to try and stay out of the frost and snow this winter.
I really need to learn how to edit my own INDI driver? not sure what software they're built with? I've never coded, probably too old to learn now too :)

commit a4eae0d3 (2021-11-06): sestosenso: Increase max allowed focuser position (Fixes #1555) (#1557)
closed 2021-11-06 by knro