indilib/indi #1559: Pegasus Astro DMFC not working with DC Motor

dennisfelczy, 2021-11-09:
I want to use PegasusAstro DMFC with a DC Motorfocus on Stellarmate

When I switch to DC Motor mode I can control the Motor via the knob and via the standalone App but not in Stellarmate/EKOS

According to the Interface documenation (http://pegasusastro.com/wp-content/uploads/2015/01/DMFC-Serial-Command-Table.pdf) the G commands should be used for relative movements. 

The indi driver seem to have implemented the M command for absolute as well as for the relative movements. (https://github.com/indilib/indi/blob/master/drivers/focuser/dmfc.cpp) 

I could imagine, that the dmfc might reject M commands when Motortype is DC

Unfortunately I don't know how to correct this in the driver and bring it to Stellarmate :-(

knro, 2021-11-10:
Please test now in StellarMate. Open the console and type in this:

`update_indi_core`

Edit your equipment profile, and set **Auto Connect** to off (uncheck it). When you start the profile, go to INDI Control Panel --> DMFC --> Select DC Motor in Main Control. After that's done, stop your profile and edit it again to enable _Auto Connect_ again. Save profile and then start again. Does it work?

closed 2021-11-10 by knro

dennisfelczy, 2021-11-10:
It works. Thank you so much for the help !!!

commit 63e0f2f6 (2021-11-10): Add preliminary support for DC focusers to DMFC. Needs testing.

Fixes #1559