indilib/indi #1560: Meridian Flip Spins OTA - iEXOS-100 - INDI Driver

wx273, 2021-11-10:
**Describe the bug**
A clear and concise description of what the bug is.  Provide a general summary of the issue in the Title above as well as affected program name like `indiserver` or `indi_toupcam_ccd`
The iEXOS-100 (PMC-Eight) mount works correctly until it executes a Meridian Flip. The mount does switch sides and does a plate solve but then continues in same direction and turns 540 degrees on both axes if the OTA doesn't impact the tripod (depending on the OTA setup, it has impacted the tripod). 

**To Reproduce**
Exact steps to reproduce the behaviour.
1. Go though regular polar alignment using EKOS and configure for MF after 0 degrees
2. Slew to target on east side of mount. 
3. Commence imaging session.
4. When mount reaches meridian observe it move from west-facing-east to east-facing-west then plate solve.
5. After plate solve mount continues moving further east-facing west until it impacts the mount
6. More info here -> https://indilib.org/forum/ekos/10621-meridian-flip-spins-ota.html

**Expected behavior**
After MF and plate solves expect mount to continue tracking target and recommence imaging session.

**Screenshots**
No screenshots but have photos of mount with power/usb cables wrapped around it.

**Desktop (please complete the following information):**
 - Ubuntu 20.04 LTS fresh install
 - Exact build process is explained here -> https://indilib.org/forum/general/10521-another-build-experience-story-ubuntu-20-04lts.html
 - Using the INDI PMC-8 driver which doesn't seem to detect which side of the pier the OTA is on

**Log Files**
Will collect log files at next session. The EKOS log showed the following:
2021-11-05T03:45:26 CCD capture aborted
2021-11-05T03:45:26 Post-flip alignment failed.
2021-11-05T03:45:13 Post-flip alignment failed. Retrying...
2021-11-05T03:45:00 Post-flip alignment failed. Retrying...
2021-11-05T03:42:56 Performing post flip re-alignment...
2021-11-05T03:42:56 Telescope completed the meridian flip.
2021-11-05T03:41:38 Received image 6 out of 50.
2021-11-05T03:41:38 Download Time: 0.71 s, New Download Time Estimate: 0.74 s.
2021-11-05T03:41:04 Capturing 30.000-second image...
2021-11-05T03:41:04 Dither complete.
2021-11-05T03:41:04 Dithering succeeded.
2021-11-05T03:41:01 Dithering...
2021-11-05T03:41:01 Received image 5 out of 50.
2021-11-05T03:41:01 Download Time: 0.72 s, New Download Time Estimate: 0.74 s.
2021-11-05T03:40:27 Capturing 30.000-second image...

knro, 2021-11-10:
@karlrees Hi Karl! is this something that the INDI driver responsible for?

karlrees, 2021-11-10:
Maybe?  I'm not entirely clear what's happening, though.  Sounds like meridian flips just fine.  So is the problem that right after the plate solve the mount is doing a goto that takes it 540 degrees the opposite direction?  Maybe the sync command is triggering what should have been a trivial goto, but the math is wrong so everything gets screwed up?

Are you sure your location and time information is right?  Are you in the northern hemisphere (if it's the southern, that hasn't been tested at all to my knowledge).

I would have thought I would have experienced something like this if it was the mount driver, but then again I'm on the Exos 2 instead of the iExos 100.  I also haven't pulled out the mount for months, and perhaps I didn't test that specific scenario since I last updated the drivers.

Not sure when I'll have time to test.  Could be a while.  I'm working two jobs and taking a class in the evenings right now.

wx273, 2021-11-21:
OK I finally have a night with a few starts to dig into this.
I managed a number of sessions with MF working correctly and the last one didn't and struck the mount.
Managed to capture the logs and will attach them here.

If I select a star just before meridian flip and center & track , then send goto to the mount in kstars, Then Capture & Solve with "Sync" set, it flips correctly at 0 degrees past, whether imaging or not. See all except last log file.

If I select a star just before meridian flip and center & track, then send goto to the mount in kstars, then Capture & Solve with "Slew to Target" set, it hits the moun
[indilog.zip](https://github.com/indilib/indi/files/7577025/indilog.zip)
t.

karlrees, 2021-11-25:
I'm not immediately seeing the problem.  It's only the last log where the problem occurred?  And only if you do Capture & Solve with Slew to Target?  I might have to pull out my scope and experiment to find the problem.  Not sure when I'll have time to do that.

wx273, 2021-11-25:
Hello Karl,
A bit more info from almost clanging the OTA into the mount again...
Imaging the Sun today from the northern hemisphere (UK). Sun low to the south and about 10 degrees east of meridian..
Mount was east facing west.
Meridian Flip was unselected.
Slew to target was checked but I wasn't doing any solving - obviously.
When I finished imaging I issued the only park command in the session and the OTA started rotating directly downwards. I saw it and issued the stop command for the mount.
Log attached. Wade past the imaging stuff by searching for the Park command.
[log_11-48-39.zip](https://github.com/indilib/indi/files/7603014/log_11-48-39.zip)
It seems to lose its way if Slew to Target is selected, even without doing any solving.

nigeldun, 2023-01-03:
Was the cause of this problem identified? I ask as I have an iexos-100 and experience the same type of problem with flips. When I’ve seen it happen it looks like it flips ok and then tries to platesolve to ensure it’s pointing at the right spot at which point it acts as if I’m in australia rather than the uk and ends up twisted and pointing at the ground.

wx273, 2023-01-04:
I'm afraid not Nigel. I gave up on this, sold the mount and bought a HEQ5 Pro.

closed 2024-04-17 by github-actions[bot]