indilib/indi #1564: Add Star Alignment option to Rainbow Astro RST-135(E) mount driver

wjdrijfhout, 2021-11-10:
The RST-135(E) mount offers an option to build a star alignment model in-mount, based on 5 or more star alignment points. With a proper in-mount model, it is possible to use this mount for decent exposures of unguided imaging. The mount model also enhances GoTo results, which can be important for small/faint targets.
However, with the current driver, the 'sync' function only sends an RA/DEC error to the mount, and it is not interpreted as a star alignment point. If you want to build a mount model, you have to do that manually. 

What I would like to have is a checkbox  in the driver to turn on 'star alignment on sync'. When ticked, a regular sync command from Ekos would then be interpreted in the mount as star alignment point. As soon as the mount has received at least 5 star alignment points, it automatically calculates this model. If such an option is available, the existing mount model functionality in Ekos can be used to just define five (or more) random alignment points, press 'run', and the model builds itself in the mount. Once the model is built, the check-box can be unchecked again if desired.

The alternative is either a fully manual method (GoTo star, center star using handpad, sync in handpad, repeat five times) or a semi-automated method (GoTo star, 'Sync & Slew to target' in ekos, sync in handpad, repeat five times). These are both annoying and time-consuming options.

I am in contact with BJ from Rainbow Astro and he is willing to share the protocol for star alignment to be included in the INDI driver. The ASCOM driver has a similar feature built in already.

sywong2000, 2021-11-17:
@wjdrijfhout Jasem has merged the code and the feature will be available in the nightly build soon. Try it out and let us know if it works okay.

wjdrijfhout, 2021-11-17:
Thanks @sywong2000 so that would be available tomorrow then. I'll certainly try it out tomorrow and let you know. Thanks so much!

jimmy-ck, 2021-11-18:
@wjdrijfhout Unfortunately it will take a little longer since there was a fundamental init missing, therefore no new Alignment tab and the Switch control are not visible. A PR fixing that including some other goodies was created yesterday: https://github.com/indilib/indi/pull/1572

wjdrijfhout, 2021-11-18:
Thanks for the update, @jimmy-ck  On standby...

wjdrijfhout, 2021-11-22:
Just curious, has this made it to the nightly build yet? I see some fixes related to this topic, but not sure yet if all is finished.

jimmy-ck, 2021-11-22:
Tbh, I can't tell since I don't get my StellarMate version to flip to nightly builds. You should notice the new driver when 2 new tabs are visible (Alignment and General Info).

jimmy-ck, 2021-11-22:
@wjdrijfhout Nightly builds should be available as far as the repos tell.

knro, 2021-11-22:
@jimmy-ck What's the issue with switching to nightly? You can also do it from the terminal:

sudo /etc/stellarmate/set-release.sh nightly

jimmy-ck, 2021-11-22:
@knro This command outputs "Release is already nightly" while StellarMate Tools is showing "stable".
Interestingly though the software update now pulls the latest Rainbow nightly. Maybe I tried it too early the last time (at least 2 days ago). And as far as I can tell [here](https://ppa.stellarmate.com/repos/apt/nightly/pool/main/libi/libindi/) nightly got updated yesterday. Thanks for pointing that out, Jasem.

sywong2000, 2021-11-23:
@knro I pull the nightly ubuntu repo and indi-full was installed as 1.9.3~202111201956~ubuntu18.04.1. Do not see the new driver tabs after installation. Building from the source shows fine.

wjdrijfhout, 2021-11-23:
Just tested the nightly build on Ubuntu 20.04. All worked as intended. This is a really great addition to the driver! Thank you so much, Stephen and Christopher! 
Just to note that the automated mount modeling tool crashed the application, but this might be due to using the CCD simulator. So for now it means manually selecting 5-6 points and run an automated goto-sync cycle on them.

knro, 2021-11-24:
Great work everyone! Can we get some documentation started for INDI site? This is an example: https://indilib.org/telescopes/astrotrac-360.html

Can someone create a new entry for this driver and submit it? You can zip the screenshot and attach them here so I can upload them. It would be great to have solid documentation there for this mount.

knro, 2021-12-16:
1. Is this issue resolved?
2. Can we get help with the documentation?

sywong2000, 2021-12-16:
@knro I think it is - I also fixed the issue on the firmware display and tested it, and made a separate pull request. I will prepare the documentation and upload.
@jimmy-ck @wjdrijfhout  see if you have anything to add on this item?

wjdrijfhout, 2021-12-22:
@sywong2000 , @knro, @jimmy-ck. As far as I can see the issue has been resolved. Tried it this week again, and works like a charm. Great improvement for using the mount unguided.
Haven't tested the automated mount model routine in Ekos again. Last time I did, it kept crashing the application. That is unrelated to the driver-issue. Also happens with simulators only.

closed 2021-12-23 by knro