indilib/indi #1571: GoTo Error: RST-235

starkgb, 2021-11-16:
**Describe the bug**
I’m running StellarMate OS 1.6.1 on a 4GB Raspberry Pi 4, and I’m seeing errors between the Pi and my RST-135 mount.  I’ll boot the mount, let it sync to gps, and then make sure that the mount is homed before I start KStars.  From within KStars, I’ll tell the mount to “GoTo” the target.  Sometimes it works, but other times I get an error.  If I do get an error, the mount moves fine on the second attempt.  Note that I’m running locally (I’ve attached a keyboard, mouse, and monitor to the Pi).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Turn on RST-135
2. Let the mount sync to gps
3. Home the mount via the hand controller
4. Start KStars
5. Start and connect via Ekos
6. Point at target in KStars, right click, select Rainbow RST-135 and then GoTo from the menu
7. Watch KStars either move to the target or throw an error (e.g., 2021-11-14T15:36:07: [ERROR] Error Slewing to JNow RA 10:09:31.1 - DEC 11:51:42.6)
8. if KStars throws an error, select the target again and watch it move correctly on the second attempt (e.g., 2021-11-14T15:36:24: [INFO] Slewing to RA: 10:09:31.1 - DE: 11:51:42.6)

**Expected behavior**
I expect the mount to slew to the target selected in KStars each and every time unless it's outside limits

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: StellarMate OS 1.6.1 and all that comes with that (I'm not sure of the underlying OS)

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

[log_08-12-28.txt](https://github.com/indilib/indi/files/7547621/log_08-12-28.txt)

knro, 2021-11-16:
@sywong2000 @jimmy-ck Any idea on this issue?

jimmy-ck, 2021-11-16:
@knro I've never observed this issue.
@starkgb Maybe a dumb sounding question: But did you try another USB cable from Stellarmate to RST-135? Reason I'm asking is because I see that in Rainbow::slewToEquatorialCoords() three serial communcation commands get fired quite quickly after each other. Maybe something get's lost there.

sywong2000, 2021-11-17:
@knro me neither
@starkgb another thing you can try is to flash the firmware again. One time I experienced an “imcomplete” flash: at the end of the firmware flashes there’re errors. But then the mount seemed to work ok and I start experiencing problem in Satellite tracking. I re-flash the firmware again and there’re no more error at the end of the flash, and the mount works normally again

sywong2000, 2021-11-17:
@starkgb did you experience the similar when you operate from the iHubo or other software?

starkgb, 2021-11-17:
Before I posted, I tried reflashing the firmware and even a different SD card.  The mount seems to work fine from the hand controller.  I've tried a few different cables, and while some seem much better than the others, I still get the occasional error.  I'm beginning to think there's a problem with the USB socket in the mount.  I've reached to the vendor about getting a replacement part, and I'm waiting to hear from him.

knro, 2021-11-24:
Thanks for the update. Will close this for now, please re-open if you can confirm the issue again.

closed 2021-11-24 by knro

starkgb, 2021-11-24:
As mentioned earlier, much of the problem seemed to be associated with the usb socket in the mount; however, I still see occasional errors with a goto after the mount has been sitting for a while.  Unfortunately, those aren’t easy to reproduce.

Thanks,
Greg

sywong2000, 2021-11-24:
Greg,
Looks like the USB socket to me. Suggest to try on other platform (e.g. ASCOM, SkySafari via wifi) and see if there're different behaviors.

starkgb, 2021-11-24:
Very well could be; I’m trying to confirm.  I’ve reached out to both the manufacture (Rainbow Astro) and the vendor (Tolga Astro) about a new circuit board, and neither have been very responsive.  I’ll try both again and hopefully one of them will respond.

Thanks again,
Greg