indilib/indi #1579: Lx200 onstep not working correctly

GrayOneAus, 2021-12-20:
Hi I am having trouble with the onstep diver not connecting correctly
my set up is a rasberry pi 4 running Stellarmate.
I have managed to get the bluetooth working so can get the unit talking to the onstep mini OK. (rfcomm1)
However when I get to the Ekos and set it to lx200 onstep it will not connect (at least not right away)
If I run say lx200 basic or lx200 teenastro and then close them. LX Onstep suddenly works and finds the rfcomm1 in my case which the bluetooth is attached.
I have noted other people have had the same issue but no answer to the problem.
I trust sombody can help with this issue

knro, 2021-12-27:
@azwing @james-lan Any ideas on handshake with serial over bluetooth? should it require any special treatment?

azwing, 2021-12-27:
@GrayOneAus
I suppose you have OnStep Mini PCB2 wih a HC05 Bluetooth module connected n lieu of the Wemos D1? Right?
Also I presume you connect to Bluetooth using the Raspberry Bluetooth.
So I imagine OnStep and Raspberry are near to each other (less than 2 m).

I never tested OnStep over Bluetooth as main connection, rather to communicate with the "OnStep App" on android .

But in principle there should be no difference with USB-Serial.
Did you setup "Serial" "/dev/rfcomm1" and "9600" in the "connection tab"?

Did you try communicate with "minicom" and issuing commands like ":GR#" and ":GU#" at least to make sure connection works?
If so, did you test running OnStep directly with device manager (out of Ekos)

What is if you connect your OnStep via USB to your Raspberry?

I have no Bluetooth module on hand but as soon I get one will make a try. 
May be you ask also on the OnStep Forum, maybe somebody else did try this set-up.

GrayOneAus, 2021-12-27:
Hi azwing
all Your assumsions are all correct
Yes I have checked communication and works fine
(in fact when I do get a connection through EKOS its fine its just the initial connection, now have a seperate profile to run lx200 basic then when its connected I switch over to lx200 onstep. this over comes the problem)
Havent tried onstep via usb two reasons have the hand controller and all the usb on pi are used up (cameras, focuser, GPS, USB Drive for files )
Thanks for your help

azwing, 2021-12-27:
So what you are describing seems to me something related to the serial link.
Could you make the following test using a smartphone with "OnStep App".
Connect with the Android App and the disconnect
Then try to connect with Ekos and see if it connects on the fly.

I just discovered that I need also a Bluetooth dongle to connect on PC side :-(
Will take a while until I can do some tests. we are in Christmas / New-year, ordering something now is quite a challenge :-)
But It would really be interesting to ask on the "Onstep Forum" https://onstep.groups.io/g/main.

rchadgray, 2022-01-03:
Check out this instructable.  I did this project and it uses RaspberryPi INDI server connecting to OnStep via bluetooth.  Worked great about a year ago.  I have not updated my Microscope software since then.  I dont know if recent updates would break it.
https://www.instructables.com/The-Micro-Scope-a-Miniture-GOTO-Telescope/

Down in the bluetooth header of this page may be some ideas.  I know he has a txt file for the rfcomm service

[Unit]
Description=RFCOMM service
After=bluetooth.service
Requires=bluetooth.service

[Service]
ExecStart=/usr/bin/rfcomm bind hci0 [mac address]

[Install]
WantedBy=multi-user.target

GrayOneAus, 2022-01-05:
Hi Greg
Thanks for the info
That covers the missing link on how to get a permanent  bluetooth connection.
That project is assume very well explained, it has also given me some idears for my next project.
I wonder if it would be possible to cool the camera with a peltier chip? to reduce the noise.

rchadgray, 2022-01-05:
My name is Chad.  Greg is the person that designed and made the Microscope.

I have wondered also about putting a peltier on the Pi HQ sensor also, but everyone that has tried has not gained much.  Everything i have read says that it is not worth it.

If you google it you will find a few people that tried.

GrayOneAus, 2022-01-05:
my apologize  chad
Thats interesting I wonder why there is no gain!

rchadgray, 2022-01-05:
No problem. I still want to try it myself. the investment in a peltier for a test is not that expensive.

Maybe there are some people out there that have had success i have not looked in a while.  I also wonder if there is a good way to get the cold to transfer through the PCB. Like we use heat sink paste to attach a heatsink to a CPU.  What should be used to get a peltier to have superior contact to the PCB for cooling?

james-lan, 2022-01-05:
Apologies for not seeing this recently. 

@GrayOneAus Which version of INDI/lx200_OnStep are you running? The current git or an earlier version. I fixed a lot of issues in what will be indi 1.9.4 related to connections. I wouldn't expect it to work well with 1.9.3. I thought that included the bluetooth connection. I just found a bluetooth dongle I have, so I'll see about testing, but any information/logs would be good. 

Board I have (not 100% sure it works) is a JY-MCU BT_BOARD v1.02

@knro The serial shouldn't require any (odd) handshakes, no, but it was having problems with the ACK that was sent prior to the last round of updates.

GrayOneAus, 2022-01-06:
Hi Chad
looking at the rear of the Pi HQ sensor the only way would be to use a THERM-A-GAP™ Dispensed Thermal Gels like T636G and may be a 3d printed holder to keep it in place which would also have the cooling fan incorporated in its feature, Some people seem to have success and the images look good though little detail how it was done. Also no temperature noted

rchadgray, 2022-01-06:
I am getting ready to redesign the camera section of my "Microscope" to add a stepper motor for focusing.  Focusing is the one part that is a real pain with this tiny telescope.  Well polar alignment is not easy when you have it on a tripod either, but manageable with a good geared tripod head.

I will check out the Therm-a-gap and see how i can incorporate the peltier. Of course then i will want to turn the cooler on/off via INDI/Ekos... hmm i see a big rabbit hole coming towards me!  LOL!

GrayOneAus, 2022-01-08:
Hi Chad
Had the same problem using canon 200mm f/l lens, had some success using a Bahtinov mask using zwo camera
polar aligment was fairly easy using APT and plate solving
the rest will be fun let me know how you go

Further to the bluetooth issue
after some playing around and research, have come to this conclusion
"Astroberry" uses the standard basic bluetooth setup (could not get it to connect to any bluetooth devices!)
"Stellermate uses Blueman & Bluez which has a nice interface/manager it connects to any bluetooth in range?
This is where i believe the onstep driver has an issue.
In frustration and a new mount i built a bluetooth dongal for my Crissy present HQE5.Skywatcher
The result it worked straight away using the EQMOD driver (stellarmate setup)
The only issue I have to first connect the bluetooth to the serial port at each start up 
Other than that it appears to work fine now only need a break in the weather to give it a go

james-lan, 2022-01-10:
Note: Try to keep things to one topic, and open another if something else isn't working. (You can reference it elsewhere.) 

_As far as I can tell all the issues I've seen are related to the port not being setup by the OS/tools._ At least with my board, I also tested copying the Handshake() from both lx200_TeenAstro, lx200telescope (which should be the one it's using), and without changes, just to make sure it's not that. 

Sometimes it had problems getting the input to it in general in the past, but after trusting it in KDE's bluetooth settings, and running: 

sudo  rfcomm bind hci0 $MAC_ADDRESS

I can't seem to make it fail now. (Which given my history with Bluetooth being a pain prior to this, is both relieving and quite frustrating because people have had issues.) Even with straight powering it down, where it's seemed to have failures before. That should be entirely equivalent to @rchadgray in the instructable, except the bind needs to be done each time. (The service is the way to go for normal use.) 

Note: My version of KStars isn't the newest, so this testing was done on a version before the connection info fix, I don't think that affected anything but not 100% sure. I'll try it again tomorrow when I'm not so tired.

knro, 2022-02-19:
Is this issue resolved?

azwing, 2022-02-19:
@knro Jasem,

On my side I could not reproduce the issue with Bluetooth so consider it as closed.

GrayOneAus, 2022-02-20:
Hi all no the problem is still there

However I think I know the problem
The bluetooth part works OK but is connected to rfcomm1

lx200 onstep I believe is looking for rfcomm0 (looking through the code)
wish there was some comments in the code or(bash)as iam no expert on this language 

this also appears to be the issue with other devices attached to the rasderry pi ie gps memery stick (some times works some times it all falls over especially if kstars crashes) only way to fix it is to reboot the system

closed 2022-06-09 by knro