indilib/indi #1580: auxiliary/pegasus_upb strange behaviour with USB_PORT_CONTROL switch vector

gnthibault, 2021-12-23:
**Describe the bug**
I am currently developing a small software to drive a telescope setup, and the UPBV2 is a central piece of my setup.
Unfortunately, I recently came across a weird behaviour with regard to the switchvector called "USB_PORT_CONTROL"

My current strategy for every switchvector, as a client, is to send the vector to the indiserver, and then wait for the corresponding light to be set to Ok (or eventually idle, but that's another story). This is somehow an acknowledgment mechanism, and the light is set to Ok, independantly of wether any or none of the switch value have actually been changed.

Anyway, this acknowledgment strategy is working pretty ok on most of the drivers I have been playing with recently, also, it is working on the "POWER_CONTROL" and "POWER_ON_BOOT" switchvector. However, when it comes to "USB_PORT_CONTROL", I noticed that the driver is never acknowledging the switch reception in the case that the sent switchvector has no difference with the current switch values.

Second issue:
In addition, I noticed that the switchvector named "USB_HUB_CONTROL" always exhibit an "Alert" value when used. I suppose there is an issue somewhere with this feature inside the indi driver, but I am not competent enough to debug this.

Is there anyway for someone to crosscheck if the behaviour can be reproduced, and if it is expected ?

Thank you in advance for your help

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run indi_pegasus_upb driver
2. Set USB_PORT_CONTROL switch vector, with some values for PORT_1, PORT_2, PORT3, PORT_4
3. Try to reproduce the step 2, i.e with the exact same switch values
3. You should notice that no acknowledgement is never sent for the second call

**Expected behavior**
I expect the driver to send a "USB_PORT_CONTROL" switchvector with a light set to "Ok" independantly of wether the latest client command did change or didn't

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Linux rockpro64 5.15.7-1-MANJARO-ARM #1 SMP PREEMPT Wed Dec 8 13:17:00 UTC 2021 aarch64 GNU/Linux
 - Version Hash ce7416338a3f99888b21c2f284aa2033718c5e6f

**Log Files**
INFO	10746.665424 sec	: Session log file /home/user/.indi/logs/2021-12-23/indi_pegasus_upb/indi_pegasus_upb_20:21:24.log
DEBUG	10746.793145 sec	: CMD <PA>
DEBUG	10746.880150 sec	: RES <UPB2:13.8:0.6:7:17.7:46:5.9:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10746.880725 sec	: CMD <PC>
DEBUG	10746.927989 sec	: RES <0.69:8.14:112.48:42737925>
DEBUG	10746.928623 sec	: CMD <SA>
DEBUG	10746.944142 sec	: RES <0:0:0:0>
DEBUG	10746.944537 sec	: CMD <DA>
DEBUG	10746.960128 sec	: RES <DA:210>
DEBUG	10747.961890 sec	: CMD <PA>
DEBUG	10748.048001 sec	: RES <UPB2:13.8:0.6:7:17.7:46:5.9:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10748.048573 sec	: CMD <PC>
DEBUG	10748.095765 sec	: RES <0.69:8.14:112.48:42739091>
DEBUG	10748.096410 sec	: CMD <SA>
DEBUG	10748.111919 sec	: RES <0:0:0:0>
DEBUG	10748.112312 sec	: CMD <DA>
DEBUG	10748.127964 sec	: RES <DA:210>
DEBUG	10749.129714 sec	: CMD <PA>
DEBUG	10749.215683 sec	: RES <UPB2:13.8:0.6:7:17.7:46:5.9:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10749.216281 sec	: CMD <PC>
DEBUG	10749.263528 sec	: RES <0.69:8.14:112.48:42740257>
DEBUG	10749.264176 sec	: CMD <SA>
DEBUG	10749.279689 sec	: RES <0:0:0:0>
DEBUG	10749.280083 sec	: CMD <DA>
DEBUG	10749.295648 sec	: RES <DA:210>
DEBUG	10750.297405 sec	: CMD <PA>
DEBUG	10750.383458 sec	: RES <UPB2:13.8:0.6:8:17.7:46:5.9:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10750.384076 sec	: CMD <PC>
DEBUG	10750.431296 sec	: RES <0.69:8.14:112.48:42741423>
DEBUG	10750.431941 sec	: CMD <SA>
DEBUG	10750.447451 sec	: RES <0:0:0:0>
DEBUG	10750.447939 sec	: CMD <DA>
DEBUG	10750.463422 sec	: RES <DA:210>
DEBUG	10751.465183 sec	: CMD <PA>
DEBUG	10751.551275 sec	: RES <UPB2:13.8:0.6:7:17.6:46:5.8:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10751.552007 sec	: CMD <PC>
DEBUG	10751.599088 sec	: RES <0.69:8.14:112.48:42742587>
[... 444 lines cut ...]
DEBUG	10807.811512 sec	: RES <DA:210>
DEBUG	10808.813233 sec	: CMD <PA>
DEBUG	10808.899667 sec	: RES <UPB2:13.8:0.6:7:17.6:46:5.8:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10808.900283 sec	: CMD <PC>
DEBUG	10808.947474 sec	: RES <0.69:8.15:112.61:42799840>
DEBUG	10808.948209 sec	: CMD <SA>
DEBUG	10808.963632 sec	: RES <0:0:0:0>
DEBUG	10808.964018 sec	: CMD <DA>
DEBUG	10808.979619 sec	: RES <DA:210>
DEBUG	10809.981372 sec	: CMD <PA>
DEBUG	10810.067443 sec	: RES <UPB2:13.8:0.6:8:17.6:46:5.8:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10810.068135 sec	: CMD <PC>
DEBUG	10810.115235 sec	: RES <0.69:8.15:112.61:42801006>
DEBUG	10810.115879 sec	: CMD <SA>
DEBUG	10810.131395 sec	: RES <0:0:0:0>
DEBUG	10810.131789 sec	: CMD <DA>
DEBUG	10810.147380 sec	: RES <DA:210>
DEBUG	10811.149201 sec	: CMD <PA>
DEBUG	10811.235141 sec	: RES <UPB2:13.8:0.6:7:17.6:46:5.8:0000:000010:0:0:0:0:0:0:0:0:0:0:0000000:6>
DEBUG	10811.235760 sec	: CMD <PC>
DEBUG	10811.282996 sec	: RES <0.69:8.15:112.61:42802171>
DEBUG	10811.283641 sec	: CMD <SA>
DEBUG	10811.299158 sec	: RES <0:0:0:0>
DEBUG	10811.299544 sec	: CMD <DA>
DEBUG	10811.315112 sec	: RES <DA:210>

knro, 2021-12-26:
Thank you for the report. You can look at the implementation of [USBControlV2SP](https://github.com/indilib/indi/blob/ce7416338a3f99888b21c2f284aa2033718c5e6f/drivers/auxiliary/pegasus_upb.cpp#L689) property and it always sends a response from what I can see either Ok or Alert depending on the result.

gnthibault, 2021-12-26:
> Thank you for the report. You can look at the implementation of [USBControlV2SP](https://github.com/indilib/indi/blob/ce7416338a3f99888b21c2f284aa2033718c5e6f/drivers/auxiliary/pegasus_upb.cpp#L689) [...]

Ok thank you so much for the analysis ! I'll soon write a driver with native C++ framework so that I'll be able to debug/check this fo rmyself in the future. 

Sorry for the inconvenience, I'll re-re-recheck my client code / debug logs then to understand what is happening. Thank you again for your valuable help

gnthibault, 2021-12-26:
Ok I tried to checkout the logs on driver side to see what was happening there, here is what I see in the erroneous case (ie, trying to set switch vector to the same value that the driver already has):

client ligne 133

With errors:
2021-12-26T11:47:01: Client 0: read newSwitchVector Pegasus UPB USB_PORT_CONTROL 
     PORT_1='On'
     PORT_2='Off'
     PORT_3='Off'
     PORT_4='Off'
     PORT_5='On'
     PORT_6='Off'
2021-12-26T11:47:01: Driver indi_pegasus_upb: queuing responsible for <newSwitchVector device='Pegasus UPB' name='USB_PORT_CONTROL'>
2021-12-26T11:47:01: Driver indi_pegasus_upb: sending msg copy 1 nq 1:
<newSwitchVector device="Pegasus UPB" name="USB_PORT_CONTROL">
    <oneSwitch name="PORT_1">
On
    </oneSwitch>
    <oneSwitch name="PORT_2">
Off
    </oneSwitch>
    <oneSwitch name="PORT_3">
Off
    </oneSwitch>
    <oneSwitch name="PORT_4">
Off
    </oneSwitch>
    <oneSwitch name="PORT_5">
On
    </oneSwitch>
    <oneSwitch name="PORT_6">
Off
    </oneSwitch>
</newSwitchVector>

With no errors:

2021-12-26T11:48:42: Client 0: read newSwitchVector Pegasus UPB USB_PORT_CONTROL 
     PORT_1='Off'
     PORT_2='Off'
     PORT_3='Off'
     PORT_4='Off'
     PORT_5='On'
     PORT_6='Off'
2021-12-26T11:48:42: Driver indi_pegasus_upb: queuing responsible for <newSwitchVector device='Pegasus UPB' name='USB_PORT_CONTROL'>
2021-12-26T11:48:42: Driver indi_pegasus_upb: sending msg copy 1 nq 1:
<newSwitchVector device="Pegasus UPB" name="USB_PORT_CONTROL">
    <oneSwitch name="PORT_1">
Off
    </oneSwitch>
    <oneSwitch name="PORT_2">
Off
    </oneSwitch>
    <oneSwitch name="PORT_3">
Off
    </oneSwitch>
    <oneSwitch name="PORT_4">
Off
    </oneSwitch>
    <oneSwitch name="PORT_5">
On
    </oneSwitch>
    <oneSwitch name="PORT_6">
Off
    </oneSwitch>
</newSwitchVector>

2021-12-26T11:48:42: Driver indi_pegasus_upb: read setSwitchVector Pegasus UPB USB_HUB_CONTROL Idle

++++++++++++++++++++++++++++++++++

2021-12-26T11:48:43: Driver indi_pegasus_upb: read setSwitchVector Pegasus UPB USB_PORT_CONTROL Ok
     PORT_1='Off'
     PORT_2='Off'
     PORT_3='Off'
     PORT_4='Off'
     PORT_5='On'
     PORT_6='Off'
2021-12-26T11:48:43: Client 0: queuing <setSwitchVector device='Pegasus UPB' name='USB_PORT_CONTROL'>
2021-12-26T11:48:43: Client 0: sending msg copy 1 nq 2:
<setNumberVector device="Pegasus UPB" name="POWER_SENSORS" state="Ok" timeout="60" timestamp="2021-12-26T11:48:43">
    <oneNumber name="SENSOR_VOLTAGE">
13.800000000000000711
    </oneNumber>
    <oneNumber name="SENSOR_CURRENT">
0.5
    </oneNumber>
    <oneNumber name="SENSOR_POWER">
7
    </oneNumber>
</setNumberVector>

2021-12-26T11:48:43: Client 0: sending msg copy 1 nq 1:
<setSwitchVector device="Pegasus UPB" name="USB_PORT_CONTROL" state="Ok" timeout="60" timestamp="2021-12-26T11:48:43">
    <oneSwitch name="PORT_1">
Off
    </oneSwitch>
    <oneSwitch name="PORT_2">
Off
    </oneSwitch>
    <oneSwitch name="PORT_3">
Off
    </oneSwitch>
    <oneSwitch name="PORT_4">
Off
    </oneSwitch>
    <oneSwitch name="PORT_5">
On
    </oneSwitch>
    <oneSwitch name="PORT_6">
Off
    </oneSwitch>
</setSwitchVector>

gnthibault, 2021-12-26:
Sorry, I cannot wrap my mind around this, it almost looks like there is an issue with the driver answering with a USB_HUB_CONTROL to a USB_PORT_CONTROL newswitch request.

Sample with erroneous behaviour:

> 2021-12-26T14:08:43: Client 0: read newSwitchVector Pegasus UPB USB_PORT_CONTROL 
>      PORT_1='On'
>      PORT_2='Off'
>      PORT_3='Off'
>      PORT_4='Off'
>      PORT_5='On'
>      PORT_6='Off'
> 2021-12-26T14:08:43: Driver indi_pegasus_upb: queuing responsible for <newSwitchVector device='Pegasus UPB' name='USB_PORT_CONTROL'>
> 2021-12-26T14:08:43: Client 5: queuing <newSwitchVector device='Pegasus UPB' name='USB_PORT_CONTROL'>
> 2021-12-26T14:08:43: Client 5: sending msg copy 2 nq 1:
> <newSwitchVector device="Pegasus UPB" name="USB_PORT_CONTROL">
>     <oneSwitch name="PORT_1">
> On
>     </oneSwitch>
>     <oneSwitch name="PORT_2">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_3">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_4">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_5">
> On
>     </oneSwitch>
>     <oneSwitch name="PORT_6">
> Off
>     </oneSwitch>
> </newSwitchVector>
> 
> 2021-12-26T14:08:43: Driver indi_pegasus_upb: sending msg copy 1 nq 1:
> <newSwitchVector device="Pegasus UPB" name="USB_PORT_CONTROL">
>     <oneSwitch name="PORT_1">
> On
>     </oneSwitch>
>     <oneSwitch name="PORT_2">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_3">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_4">
> Off
>     </oneSwitch>
>     <oneSwitch name="PORT_5">
> On
>     </oneSwitch>
>     <oneSwitch name="PORT_6">
> Off
>     </oneSwitch>
> </newSwitchVector>
> 
> 2021-12-26T14:08:43: Driver indi_pegasus_upb: read setSwitchVector Pegasus UPB USB_HUB_CONTROL Idle
>  INDI_ENABLED='On'
>  INDI_DISABLED='Off'
>

gnthibault, 2021-12-26:
From what I understand from the code, there might be an issue at this exact line:
https://github.com/indilib/indi/blob/ce7416338a3f99888b21c2f284aa2033718c5e6f/drivers/auxiliary/pegasus_upb.cpp#L717

I think that the 
`            IDSetSwitch(&USBControlSP, nullptr);
`Should be replaced by a
`            IDSetSwitch(&USBControlV2SP, nullptr);
`

To me it looks like a copy paste kind of mistake, but I am not 100% sure, @knro  can you please quickly check ? thank you in advance for your help

gnthibault, 2021-12-26:
Ok problem fixed on my side thanks to this fix, I will try to create new PR

gnthibault, 2021-12-26:
Note: this fix is enough to get me on track for my developments, but it is still not clear what is happening with USB_HUB_CONTROL on the UPB_V2 (I think somehow this was used in V1, but I don't understand what happened there, and what should be the expected behaviour for V2).
I don't need this other feature for now anyway, lets put it aside for now.

gnthibault, 2021-12-26:
https://github.com/indilib/indi/blob/ce7416338a3f99888b21c2f284aa2033718c5e6f/drivers/auxiliary/pegasus_upb.cpp#L1227

To me it looks like there is a weird behaviour when trying to set some values equal to the previous value held by the upbv2. If someone from Pegasus astro can give it a looks and crosscheck this code, that would be great.

gnthibault, 2021-12-26:
Ok finally found the issues, fully fixed on my setup with this PR: https://github.com/indilib/indi/pull/1581

closed 2021-12-27 by knro