indilib/indi #1597: 1.9.4: by defauld cmake builds and installs static libraries

kloczek, 2022-01-17:
**Describe the bug**
Looks like by default cmake builds and installs shared and static libraries and it is not popssible to disable that on cmake run to configure source tree:
```console
[tkloczko@ss-desktop lib64]$ ls -l
total 2556
drwxr-xr-x 1 tkloczko tkloczko      22 Jan 17 10:52 indi
-rw-r--r-- 1 tkloczko tkloczko   28194 Jan 17 10:52 libindiAlignmentClient.a
lrwxrwxrwx 1 tkloczko tkloczko      27 Jan 17 10:52 libindiAlignmentDriver.so -> libindiAlignmentDriver.so.1
lrwxrwxrwx 1 tkloczko tkloczko      31 Jan 17 10:52 libindiAlignmentDriver.so.1 -> libindiAlignmentDriver.so.1.9.4
-rwxr-xr-x 1 tkloczko tkloczko  123448 Jan 17 10:52 libindiAlignmentDriver.so.1.9.4
-rw-r--r-- 1 tkloczko tkloczko  123570 Jan 17 10:52 libindiclient.a
-rw-r--r-- 1 tkloczko tkloczko  130500 Jan 17 10:52 libindiclientqt.a
-rw-r--r-- 1 tkloczko tkloczko  602748 Jan 17 10:52 libindidriver.a
lrwxrwxrwx 1 tkloczko tkloczko      18 Jan 17 10:52 libindidriver.so -> libindidriver.so.1
lrwxrwxrwx 1 tkloczko tkloczko      22 Jan 17 10:52 libindidriver.so.1 -> libindidriver.so.1.9.4
-rwxr-xr-x 1 tkloczko tkloczko 1451440 Jan 17 10:52 libindidriver.so.1.9.4
lrwxrwxrwx 1 tkloczko tkloczko      17 Jan 17 10:52 libindilx200.so -> libindilx200.so.1
lrwxrwxrwx 1 tkloczko tkloczko      21 Jan 17 10:52 libindilx200.so.1 -> libindilx200.so.1.9.4
-rwxr-xr-x 1 tkloczko tkloczko  114872 Jan 17 10:52 libindilx200.so.1.9.4
drwxr-xr-x 1 tkloczko tkloczko      20 Jan 17 10:52 pkgconfig
```

**Desktop (please complete the following information):**
 - OS: Linux x86/64
 - Version 1.9.4

knro, 2022-01-17:
You want to build only shared?

kloczek, 2022-01-17:
> You want to build only shared?

Yes 😄 
Using static libraries is always asking for troubles in case of some ABI changes or be forced to rebuild other binaries when new version of the library arrives.

knro, 2022-01-17:
I suppose we can add an option BUILD_STATIC_LIBS that's ON by default and can be turned OFF from cmake

kloczek, 2022-01-17:
That would be awesome 😄

knro, 2022-11-22:
I think this should be fixed once !1767 is merged.

pawel-soja, 2022-11-22:
Sure, I can add that option. I'd love to get rid of the static libraries, but I can't.

Please note that not all classes are safe to use as shared libraries.
I am refactoring `INDI Core` libraries so that during future development of libraries, the size of the declared class in the header file does not change (D_PTR mechanism). This approach will avoid recompilation of applications that use `INDI Core` libraries.

Unfortunately, most classes do not have such a mechanism and if there is a change related to the size of the class, the whole thing must be recompiled (along with the program that uses the libraries).
Otherwise, the program will allocate an incorrect amount of memory and perform incorrect initializations.

Currently, static libraries that are compiled into applications are immune to library changes, with the disadvantage that a recompile of the entire library is required when changing the library.

If @knro introduces the correct version numbering adequate to the changes, it will be possible to introduce shared libraries earlier.

pawel-soja, 2022-11-28:
@kloczek From now on you can use the options provided by CMakeLists.txt

https://github.com/indilib/indi/blob/a8a5d71667757615607ca79e31d989f4540d7c4f/CMakeLists.txt#L95-L96

helper libraries can be built the old way so as not to break compatibility.
If you have any questions/problems feel free to ask.

closed 2022-11-28 by pawel-soja

kloczek, 2022-11-28:
I know aboutr that.
Issue is that it does not make any sense to build and install static libraries.

pawel-soja, 2022-11-28:
Installation of static libraries is required to properly build e.g. `KStars`. The use of shared libraries exposes the program to crash if shared libraries are replaced.

So to maintain backward compatibility, building only shared libraries is done by calling `cmake` with the appropriate arguments.

I suggest from version `2.0.0` to name static libraries differently, e.g. `libindidriverstatic.a` to have a compromise over current solutions.