indilib/indi #1600: Update documentation

paolostivanin, 2022-01-18:
The documentation at https://indilib.org/get-indi/opensuse.html should be updated as follow:

- 15.2 should be removed, as it went EOL at the end of December and I'm no longer building packages for that codestream
- 15.3 is the stable release
- 15.4 is the beta release

```
sudo zypper ar -f https://download.opensuse.org/repositories/Application:/Astrophotography/15.4/ Astrophotography_Software
```

If you can make me editor of that page, I'll gladly keep it updated!

Thanks :)

knro, 2022-01-20:
You have access now, can you please try?

paolostivanin, 2022-01-20:
Hmm I cannot find any "edit" button

knro, 2022-01-20:
Looks like you need to access administrator interface, can you login to indilib.org/administrator then components --> jdownloads ?

paolostivanin, 2022-01-20:
I can. I see only "control panel" and "configuration" though, and both give me a "An error has occurred." when I click on them.

knro, 2022-07-21:
Sorry, I can't believe I didn't response to this sooner. I don't recall if this issue is resolved or not yet?

paolostivanin, 2022-07-21:
> I can. I see only "control panel" and "configuration" though, and both give me a "An error has occurred." when I click on them.

Hi! Nope, same error as mentioned above!

knro, 2022-07-21:
I can confirm the issue.. unfortunately, I failed to fix the issue.. someone who know Joomla better than I do can perhaps figure out a better solutoin.

closed 2022-07-21 by knro