indilib/indi #1604: LX200 does not move

joergto, 2022-01-25:
LX200 compatible mount does not move. 

the driver sends the command ":SdsDD:MM#" (set DEC)
according to the meade original lx200 docs it should be :SdsDD*MM#

please see www.meade.com/support/LX200CommandSet.pdf

problem is in
lx200driver.cpp
int setObjectDEC(int fd, double dec)

knro, 2022-01-25:
@d33psky is this a regression when using short format?

d33psky, 2022-01-25:
setObjectDEC handles 3 formats of which the LX200_EQ_SHORT_FORMAT implements the :SdsDD*MM#
So if joergto is seeing :SdsDD:MM# that is very weird as there is no code that can send that because the other two formats are LX200_EQ_LONG_FORMAT with :SdsDD:MM:SS# and LX200_EQ_LONGER_FORMAT with :SdsDD:MM:SS.ss#

For reference the Meade LX200CommandSet.pdf has 
```
:SdsDD*MM#
Set target object declination to sDD*MM or sDD*MM:SS depending on the current precision setting
Returns:
1 - Dec Accepted
0 – Dec invalid
```

And the 10Micron manual has : 
```
:SdsDD*MM# or :SdsDD*MM:SS# or :Sd sDD*MM:SS.S#
Set target object declination to sDD*MM (sign, degrees, arcminutes), sDD*MM:SS
(sign, degrees, arcminutes, arcseconds) or sDD*MM:SS.S (sign, degrees, arcminutes,
arcseconds and tenths of arcsecond)
Returns:
0 invalid
1 valid
```

and on the * specifically : 

```The meaning of the asterisk (*) character changes accordingly to the emulation and precision mode
set:
Emulation and precision                                  Meaning of the asterisk (*) character
LX200 emulation, low or high precision           ascii code 223 (0xDF)
Extended emulation, low or high precision      ascii code 42 (0x2A) (ASCII asterisk "*")
Any emulation, ultra precision                         Not used in communications from the mount to the controller, both 0xDF and 0x2A accepted in communications from the controller to the mount.
```

I have never seen the ascii code 223 (0xDF) mentioned there. But the LX200_EQ_LONG_FORMAT and LX200_EQ_LONGER_FORMAT are both wrong as they use : between DD and MM which should be a *

I'll make a PR to change LX200_EQ_LONG_FORMAT and LX200_EQ_LONGER_FORMAT.

-- Hans

ps. Joergto: with which mount do you get this error ?

d33psky, 2022-01-25:
Yes that too might cause this issue. But that space has been there for ages.
The Meade spec sadly does not specify the sign, like whether a + is needed or is optional.
We can remove the space, but then we should remove it for setObjectRA :Sr as well.
I'll add it to the PR :)

joergto, 2022-01-25:
@Hans, it is a mid 90's LX200 10" with Firmware 3.20. I've reprogrammed the driver and it works with * whether space or not.
- Joerg

d33psky, 2022-01-26:
@knro ps. I (still) lack rights to close a bug.

knro, 2022-01-26:
@d33psky I added you to the triage team now

closed 2022-01-26 by d33psky