indilib/indi #1608: error: call to 'abs' is ambiguous

ryandesign, 2022-01-29:
**Describe the bug**
indi 1.9.4 builds on macOS 10.15.7 and later using the version of Apple Clang included with Xcode 12.4 and later, but does not build on macOS 10.15.7 and earlier using the Apple Clang from Xcode 11.7 and earlier. The error is:

```
drivers/rotator/deepskydad_fr1.cpp:280:9: error: call to 'abs' is ambiguous
    if (std::abs(motorPositionDouble - GotoRotatorN[0].value) > 0.01 || GotoRotatorNP.s != motionState)
        ^~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdlib.h:132:6: note: candidate function
int      abs(int) __pure2;
         ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/stdlib.h:110:44: note: candidate function
inline _LIBCPP_INLINE_VISIBILITY long      abs(     long __x) _NOEXCEPT {return  labs(__x);}
                                           ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/stdlib.h:112:44: note: candidate function
inline _LIBCPP_INLINE_VISIBILITY long long abs(long long __x) _NOEXCEPT {return llabs(__x);}
                                           ^
```

So, there are versions of `std::abs` for `int`, `long`, and `long long`, but `motorPositionDouble` is a `double`. Did you mean to use `std::fabs`?

**To Reproduce**
Exact steps to reproduce the behavior.
1. compile indi 1.9.4 on macOS 10.15.7 or earlier using Apple Clang from Xcode 11.7 or earlier.

**Expected behavior**
successful compilation

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7
 - Version 1.9.4

**Log Files**
Here is a [build log](https://build.macports.org/builders/ports-10.15_x86_64-builder/builds/89321/steps/install-port/logs/stdio) from macOS 10.5.7 with Apple Clang from Xcode 11.7.

rlancaste, 2022-01-29:
So I didn't write this driver and I can't test it since I don't have a rotator, my MacOS system is newer than the one you described, and I didn't write the driver.  But I can try changing it to std::fabs and you could try it out.  And we could ask the author of the driver if this is a good change.

rlancaste, 2022-01-29:
It looks to me like that would be appropriate based on the code it is applied to.

rlancaste, 2022-01-29:
Wait, no, I think when I fixed this in the other drivers, I had added an include

Try this: #include <cstdlib> at the top of the file

rlancaste, 2022-01-29:
#include <cstdlib>

rlancaste, 2022-01-29:
`#include <cstdlib>`

rlancaste, 2022-01-29:
Sorry for the multiple posts, the forum was removing what I wanted to include!

ryandesign, 2022-01-30:
Testing on macOS 10.14, adding `#include <cstdlib>` didn't help but adding `#include <cmath>` did!

rlancaste, 2022-01-30:
Gotcha, I knew it was one of those.

rlancaste, 2022-01-30:
Here you go, I submitted a pull request.  https://github.com/indilib/indi/pull/1611

closed 2022-02-19 by knro