indilib/indi #1610: Celestron SCT Focuser falls over on MacOs

amartinwest, 2022-01-30:
I had setup a test profile with two xwo cameras and the CST focuser. Started the calibration but upon completion? it fell over

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               kstars [95866]
Path:                  /Applications/KStars.app/Contents/MacOS/kstars
Identifier:            
Version:               3.5.6 (3.5.6)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-01-30 15:13:05.3410 +0000
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Bridge OS Version:     6.1 (19P647)
Anonymous UUID:        5387C3A9-4DA8-A8A4-14E3-A54C57A383B0

Sleep/Wake UUID:       5826B637-812D-4918-8FDE-8791A42B30E9

Time Awake Since Boot: 390000 seconds
Time Since Wake:       4595 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       UNKNOWN_0xD at 0x0000000000000000
Exception Codes:       0x000000000000000d, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [95866]

VM Region Info: 0 is not in any region.  Bytes before following region: 4443267072
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      108d6e000-109b4a000    [ 13.9M] r-x/r-x SM=COW  .../MacOS/kstars

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   kstars                        	       0x108f7f31d 0x108d6e000 + 2167581
1   kstars                        	       0x108ed2ded 0x108d6e000 + 1461741
2   kstars                        	       0x108ee68b3 0x108d6e000 + 1542323
3   QtCore                        	       0x10d8fd9c4 0x10d6e8000 + 2185668
4   kstars                        	       0x108dc9d17 0x108d6e000 + 376087
5   kstars                        	       0x108e3dc8a 0x108d6e000 + 851082
6   kstars                        	       0x108e4b54b 0x108d6e000 + 906571
7   kstars                        	       0x108e4d583 0x108d6e000 + 914819
8   QtCore                        	       0x10d8f5d39 QObject::event(QEvent*) + 745
9   QtWidgets                     	       0x10cbd4586 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
10  QtWidgets                     	       0x10cbd5940 QApplication::notify(QObject*, QEvent*) + 480
11  QtCore                        	       0x10d8cd117 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 167
12  QtCore                        	       0x10d8ce228 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 792
13  libqcocoa.dylib               	       0x10e2d26f8 0x10e297000 + 243448
14  libqcocoa.dylib               	       0x10e2d2e18 0x10e297000 + 245272
15  CoreFoundation                	    0x7ff80af368fd __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
16  CoreFoundation                	    0x7ff80af36865 __CFRunLoopDoSource0 + 180
17  CoreFoundation                	    0x7ff80af365e4 __CFRunLoopDoSources0 + 242
18  CoreFoundation                	    0x7ff80af3501b __CFRunLoopRun + 893
19  CoreFoundation                	    0x7ff80af345dd CFRunLoopRunSpecific + 563
20  HIToolbox                     	    0x7ff813b674f1 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	    0x7ff813b67247 ReceiveNextEventCommon + 587
22  HIToolbox                     	    0x7ff813b66fe5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
23  AppKit                        	    0x7ff80d963b4c _DPSNextEvent + 886
24  AppKit                        	    0x7ff80d9621b8 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1411
25  AppKit                        	    0x7ff80d9545a9 -[NSApplication run] + 586
[... 561 lines cut ...]
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:MacOS
 - Version12.1
[log_15-12-11.txt.zip](https://github.com/indilib/indi/files/7966234/log_15-12-11.txt.zip)

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

lglina, 2022-02-11:
@amartinwest I've had the same problem, and found there's a memory error in `drivers/focuser/celestron.cpp`. I have a fix. Submitting a PR now. Hopefully will fix your issue.

lglina, 2022-02-11:
cc @rlancaste

rlancaste, 2022-02-11:
I just brought this to Jasem's attention.  If it gets fixed I can see what I can do about building you another dmg.  I want them to be available on the kde binary factory, and I thought I had that working, but I ran into some issues with it this week.  So if push comes to shove, I could build another dmg.

closed 2022-02-11 by knro

amartinwest, 2022-02-11:
Thank you

regards Martin West
skype:amartinwest
mob:07525424513