indilib/indi #1615: Skywatcher Alt Az connection fail after upgrade (4gb pi 4)

saviourbot, 2022-02-03:
### Discussed in https://github.com/indilib/indi/discussions/1614

<div type='discussions-op-text'>

<sup>Originally posted by **saviourbot** February  3, 2022</sup>
After an upgrade my Alt Az will no longer connect (Debug below) I dont want to raise it as an issue if it is just something I am doing wrong has there recently been a change in InitMount() ?

The mount connests as a USB device to the PI:
`astroberry@astroberry:~ $ ls -l /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0
lrwxrwxrwx 1 root root 13 Feb  3 10:26 /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0 -> ../../ttyUSB0
`

Debug output shows no connect - Handshake failed:
INFO	0.030534 sec	: Session log file /home/astroberry/.indi/logs/2022-02-03/indi_skywatcherAltAzMount/indi_skywatcherAltAzMount_10:51:52.log
DEBUG	0.121657 sec	: Connecting to /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0 @ 9600
DEBUG	0.126972 sec	: Port FD 3
DEBUG	0.127087 sec	: Connection successful, attempting handshake...
SCOPE	0.127124 sec	: SkywatcherAPIMount::Handshake
SCOPE	0.127159 sec	: CheckIfDCMotor
SCOPE	1.129368 sec	: CMD <e1>
SCOPE	1.133705 sec	: RES <e1>
SCOPE	1.133787 sec	: Mount type not supported. 0
SCOPE	1.133817 sec	: SkywatcherAPIMount::Handshake - Result: 0
DEBUG	1.133846 sec	: Handshake failed.

`bool SkywatcherAltAzSimple::Handshake()
{
    DEBUG(DBG_SCOPE, "SkywatcherAltAzSimple::Handshake");
    SetSerialPort(PortFD);

    Connection::Interface *activeConnection = getActiveConnection();
    if (!activeConnection->name().compare("CONNECTION_TCP"))
    {
        tty_set_generic_udp_format(1);
    }

    bool Result = InitMount();

    if (getActiveConnection() == serialConnection)
    {
        SerialPortName = serialConnection->port();
    }
    else
    {
        SerialPortName = "";
    }

    RecoverAfterReconnection = false;
    DEBUGF(DBG_SCOPE, "SkywatcherAltAzSimple::Handshake - Result: %d", Result);
    return Result;
}`

It has been working fantastic but since an apt upgrade not so much. 

Thanks in advance
Gus

Update: I did a fresh build of the astroberry 2.0.4 build and it worked. Then I did an apt update && apt upgrade and I got the error again.

So, it appears to be a recent change causing the problem.</div>

ckuethe, 2022-07-21:
I'm having the same issue using the Synscan USB Dongle and an [Orion Starseeker III](https://www.telescope.com/Orion-StarSeeker-III-90mm-GoTo-Mak-Cass-Telescope/p/106368.uts).

The cables are all good since I can connect with either a terminal emulator (`minicom`) or the skywatcher motor controller firmware updater and query the motor controller version:

```
:e1
=020C83                                                                   
```

Yep, version 2.12, product code 0x83.

I know the mount is ok since I can drive it around with the hand controller. When I try connect over the serial dongle, the connection fails. Not sure where that `RES <e1>` is coming from, since my terminal emulator reads properly...

```
DEBUG	40.447259 sec	: Connecting to /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_D-if00-port0 @ 9600
DEBUG	40.448420 sec	: Port FD 3
DEBUG	40.448461 sec	: Connection successful, attempting handshake...
SCOPE	40.448474 sec	: SkywatcherAPIMount::Handshake
SCOPE	40.448484 sec	: CheckIfDCMotor
SCOPE	41.452215 sec	: CMD <e1>
SCOPE	41.456862 sec	: RES <e1>
SCOPE	41.456937 sec	: Mount type not supported. 0
SCOPE	41.456948 sec	: SkywatcherAPIMount::Handshake - Result: 0
DEBUG	41.456956 sec	: Handshake failed.
```

Strangely, connecting over the wifi adapter works.

```
INFO	0.004820 sec	: Session log file /home/ckuethe/.indi/logs/2022-07-21/indi_skywatcherAltAzMount/indi_skywatcherAltAzMount_03:20:53.log
DEBUG	8.752622 sec	: Toggle Debug Level -- Alignment Subsystem
DEBUG	9.355949 sec	: Toggle Logging Level -- Alignment Subsystem
DEBUG	10.073676 sec	: Toggle Debug Level -- Scope Verbose
DEBUG	10.657918 sec	: Toggle Logging Level -- Scope Verbose
DEBUG	12.829450 sec	: Configuration successfully saved.
INFO	46.892157 sec	: Connecting to 192.168.0.110@11880 ...
DEBUG	46.892259 sec	: Connection to 192.168.0.110@11880 is successful, attempting handshake...
SCOPE	46.892266 sec	: SkywatcherAPIMount::Handshake
SCOPE	46.892269 sec	: CheckIfDCMotor
SCOPE	48.901781 sec	: CMD <e1>
SCOPE	48.922328 sec	: RES <020C83>
SCOPE	48.922407 sec	: GetMicrostepsPerRevolution
SCOPE	48.922420 sec	: CMD <a1>
SCOPE	49.185774 sec	: RES <972B16>
SCOPE	49.185858 sec	: Axis 0: 2643.561111 microsteps/degree, 0.734323 microsteps/arcsec
SCOPE	49.185871 sec	: GetMicrostepsPerRevolution
SCOPE	49.185879 sec	: CMD <a2>
SCOPE	49.449995 sec	: RES <972B16>
SCOPE	49.450076 sec	: Axis 1: 2643.561111 microsteps/degree, 0.734323 microsteps/arcsec
SCOPE	49.450089 sec	: GetStepperClockFrequency
SCOPE	49.450098 sec	: CMD <b1>
SCOPE	49.717090 sec	: RES <4B4C00>
SCOPE	49.717169 sec	: GetStepperClockFrequency
SCOPE	49.717182 sec	: CMD <b2>
SCOPE	49.740317 sec	: RES <4B4C00>
SCOPE	49.740388 sec	: CMD <g1>
SCOPE	49.760485 sec	: RES <20>
SCOPE	49.760572 sec	: CMD <g2>
SCOPE	49.777688 sec	: RES <20>
SCOPE	49.777772 sec	: GetMicrostepsPerWormRevolution
SCOPE	49.777787 sec	: CMD <s1>
SCOPE	49.908328 sec	: RES <0>
ERROR	49.908407 sec	: Mount error: Unknown command
SCOPE	49.908419 sec	: GetMicrostepsPerWormRevolution
SCOPE	49.908428 sec	: CMD <s2>
SCOPE	49.923131 sec	: RES <0>
ERROR	49.923225 sec	: Mount error: Unknown command
SCOPE	49.923242 sec	: CMD <f1>
SCOPE	49.941467 sec	: RES <003>
SCOPE	49.941512 sec	: CMD <f2>
[... 44 lines cut ...]
:a2                                                                             
=972B16                                                                         
:b1                                                                             
=4B4C00                                                                         
:b2                                                                             
=4B4C00                                                                         
:g1                                                                             
=20                                                                             
:g2                                                                             
=20                                                                             
:s1                                                                             
!0                                                                              
:s2                                                                             
!0                                                                              
:f1                                                                             
=003                                                                            
:f2                                                                             
=003
:j1                                                                             
=000080                                                                         
:j2                                                                             
=000080                                                                         
```

And of course I can use the mount control tab in Ekos to slew the mount around.

knro, 2022-07-21:
Please test against master GIT if you can as there has been fixes pushed for SkyWatcher API.

ckuethe, 2022-07-22:
Yep, that works now.

I found some bugs in the build system. Look for patches later this evening.

closed 2022-07-22 by knro