indilib/indi #1616: CRIT message INDI driver "indi_rigel_dome" crashed!

240volts, 2022-02-03:
**Describe the bug**
The indi_rigel_dome driver crashes Kstars when running EKOS for the first time after powering up Pi4 or after a reboot. It does not crash if Kstars is run again without a hard reboot.

**To Reproduce**
1. Fresh flash of Astroberry version 2.0.3
2. Configure system for locale settings etc.
3. Reboot system
3. Run Kstars
4. Toggle Ekos
5. Create a new profile with minimum devices. Pulsar Dome and ZWO camera.
6. Run Ekos
7. dome driver crashes Kstars
8. Run Kstars again (don't reboot)
9. Toggle Ekos
10. Run Ekos with minimum profile selected.
10. Existing instance of Indi server is detected. Select the re-start indi option.
11. Ekos starts with no dome driver crash this time.

**Expected behavior**
Ekos should start first time without crashing.

**Screenshots**
rigel_dome_bug.jpg  - shows Indi Control panel and system port info after second successful start. 

**Desktop (please complete the following information):**
Raspberry Pi4. Astroberry version 2.0.3

**Log Files**
Running EKOS - first time with crash
[log_18-28-00.txt](https://github.com/indilib/indi/files/7997776/log_18-28-00.txt)
Running EKOS - next time no crash
[log_18-40-18.txt](https://github.com/indilib/indi/files/7997777/log_18-40-18.txt)
[image]

240volts, 2022-02-03:
I have noticed https://github.com/indilib/indi/pull/1596 may address this issue.

nouspiro, 2022-02-03:
You can try open that serial line with minicom or screen program and write ANGLE command. It should return current angle. Then close that serial program and Kstars should start without crash.

240volts, 2022-02-03:
Hi,

I amended the bug report as I found #1596 change to rigel_dome.cpp which may fix the issue. This change sends a dummy command at the start of the HandShake() method to the dome controller to flush the serial line.  Writing the ANGLE command is the same idea I guess but as a manual workaround.  Trying to workout how to copy the rigel dome driver to a 3rd-party driver and compile as a temporary fix until this bug fix gets included in an Astroberry update. This seems a better way but this is a learning curve for me?

 

Regards

 

David Bennett

nouspiro, 2022-02-03:
Yes manual workaround and test if it same bug.

240volts, 2022-02-04:
Hi,

I used minicom with a baud rate of 115200 and turned hardware control off. Typed ANGLE then return and it responded with E,1. Sent ANGLE again and got back 17.7.  Actually the rotator angle was 0.0.  I probably didn’t configure minicom correctly but pleased to say EKOS ran without crashing!

 

So I think it’s the same bug.

 

Not sure how to proceed from here?

 

Best regards

 

David

nouspiro, 2022-02-04:
Either wait for update or try compile INDI yourself.

240volts, 2022-02-04:
I am sorry I don’t know your name but thanks for your help.  I do find understanding who are the skilled developers behind Indi a bit of a mystery. 

I have cloned 3rd party drivers with the idea of copying the rigel_dome driver  and using my own driver until the updated one is published.  The code fix is trivial but compiling is harder. Currently running cmake and trying to resolve all the missing stuff like CFITSIO.  I want to be able to run astroberry with this dome driver fixed so I can move onto the second serious bug I have.

The second bug causes kstars to crash out.  It happens at various times and quite often when after completing an alignment using plate solving and then moving to the camera tab and taking the next image it bombs out after downloading the image.  I have seen something in the log about threads but not had time to test properly. I wanted to get rid of the first bug before testing and reporting on the second.

 

Best regards

 

David

240volts, 2022-02-07:
Hi there. Just reporting I cloned drivers and 3rd-party drivers from github and compiled my own 3rd party driver using the code from the driver with the fixed handshake method.   I was out last night observing and pleased to say had no crashes.

 

Best regards

David Bennett

closed 2022-02-19 by knro