indilib/indi #1621: Celestron Driver causing CGE error 16/17

Swizerlan, 2022-02-10:
When I issue a goto command from Asi Air Plus to my cge via nexstar + usb port it causes the hc to say error 16 and 17. Then it slews until it hits the motor stop. I fo not have these issues if I use a pc through the usb port via cpwi and nexremote.

**To Reproduce**

Initialize cge with asi air off, perform wuick align.
Plug cge nexstar+ usb up to asi air usb out. 
select the celestron cge driver, connect.

issue a goto command from asi air, this immediately triggers error 16 or 17.

Mc firmware 5.20
Hc firmware 5.31.9200

**Expected behavior**
Commands sent from asi air using celestron indy driver should issue goto and plate solves without throwing error 26&17

knro, 2022-02-10:
Hmmm I don't think a driver can cause that. These errors are usually from HC communication issues.

Swizerlan, 2022-02-10:
> Hmmm I don't think a driver can cause that. These errors are usually from HC communication issues.

I have tried 2 different HC’s with identical results. It only happens when I use asi air. I need to test the indi drivers via a pc connection tonight. I will have to load up linux though

knro, 2022-11-22:
Any update on the PC tests?

Swizerlan, 2022-11-22:
Yes and I apologize for not closing this out. A usb ground loop isolator between asi air and the nexstar controler fixed the problem

closed 2022-11-22 by Swizerlan