indilib/indi #1628: Primaluce Arco

ChristianHen, 2022-02-19:
**Did you make sure there is no existing driver for your device? Please check [supported devices portal](https://indilib.org/devices].**
Please confirm you verified no drivers exist for your device.

YES. The Arco is attached to the ESATTO. There is an existing Driver for the ESATTO, which probably can be modified to support the Arco as well (?)

**Provide link to product page**
Provide URL to the product description and details.

https://eu.primalucelab.com/esatto-3/esatto-3-robotic-microfocuser.html

**Describe device family**
All INDI devices fall under one or more of the following categories:
* Mounts
* Cameras (CCDs + CMOS)
* Focusers
* Filter Wheels
* Domes
* Weather Stations
* Adaptive Optics
* Receivers
* Spectrographs
* Auxiliary (Anything else that does not fit above).
FOCUSERS

**Provide device communication protocol**
All devices use some sort of communication language between them and the host system. This could be called *Command Set* or *SDK*. It is best to provide a link to a document or PDF with such details from the manufacutrer. It needs to include the following information:
* Communication Method: Serial, USB, TCP...etc. along with the connection parameters.
* Communication Protocol: Detailed description of the language used to communicate with the device along with examples if available.
*
Serial; I dont know the command set but i can monitor the serial port and see the commands that make the Arco move.
e.g. to make Arco move to an abs position of 20 degrees, the command is

{"req":{"cmd":{"MOT2":{"GOTO":{"DEG":20.0002}}}}}

which is very similar to the GOTO commands in the ESATTO driver

**Hardware Sample**
Would it be possible to send the developer a sample hardware unit to develop the driver? If that's not possible, would it be possible to setup a remote environment with the devie attached?

A remote environment is possible.

ChristianHen, 2022-02-20:
This is what I could find out so far about the communiaction with ARCO. The following requests and responses are being sent via the serial port in the respective situations:

DISABLE ARCO
{"req":{"set": {"ARCO": 0}}}
{"res":{"set":{"ARCO":"done"}}}

ENABLE ARCO
{"req":{"set": {"ARCO": 1}}}
{"res":{"set":{"ARCO":"done"}}}

REPEATED STANDARD STATUS COMMUNICATION
{"req":{"get": {"ARCO":"","MOT1":""}}}}} //(2x --> Why?)
{"res":{"get":{"ARCO":1,"MOT1":{"ABS_POS":0,"EL_POS":0,"MARK":0,"SPEED":0,"ACC":70,"DEC":70,"MAX_SPEED":65,"MIN_SPEED":0,"TVAL_HOLD":2,"TVAL_RUN":18,"TVAL_ACC":18,"TVAL_DEC":18,"T_FAST":25,"TON_MIN":41,"TOFF_MIN":41,"FS_SPD":39,"OCD_TH":2,"STEP_MODE":4,"CONFIG":11904,"ALARM":255,"STATUS":{"HIZ":1,"UVLO":0,"TH_SD":0,"TH_WRN":0,"OCD":0,"WCMD":0,"NOPCMD":0,"BUSY":0,"DIR":"In","MST":"stop"},"SUBMODEL":"ESATTO3","NTC_T":"30.28","BKLASH":0,"CAL_STATUS":"stop","CAL_MAXPOS":731000,"CAL_BKLASH":149,"CAL_MIN_LO":0,"CAL_MIN_HI":19364,"CAL_MAX_LO":685087,"CAL_MAX_HI":731000,"CAL_OFFMIN":19364,"HSENDET":1,"CAL_MINPOS":0,"CAL_HOMEP":1000,"CAL_NUMOF":2,"CAL_DIR":"normal","LASTDIR":0,"HOLDCURR_STATUS":0,"COMPENSATION_POS":0.00000,"POSITION":0.00000,"POSITION_MEC":0.00000}}}}

{"req":{"get": {"CALRESTART": ""}}} (2x --> why?)
{"res":{"get":{"CALRESTART":{"MOT1":0,"MOT2":1}}}}		//(="ARCO-MOTOR needs recalibration, ESATTO Motor is ok!" (Request was sent after Disable ARCO and reENABLE Arco!)

{"req":{"get": {"ARCO_LEDBLINKSTATUS":""}}}			//(Only occurs when CALRESTART = 1 for MOT2, i.e. when ARCO needs recal))
{"res":{"get":{"ARCO_LEDBLINKSTATUS":"CalRestart_MaxPosEqualMinPos"}}}

{"req":{"get":{"EXT_T":""}}}
{"res":{"get":{"EXT_T":"-127.00"}}}

{"req":{"get":{"VIN_12V":""}}}
{"res":{"get":{"VIN_12V":"12.45"}}}

{"req":{"get":{"VIN_USB":""}}}
{"res":{"get":{"VIN_USB":"5.16"}}}

{"req":{"get":{"MOT2":{"ABS_POS":"DEG","COMPENSATION_POS":"DEG","POSITION":"DEG"}}}}
{"res":{"get":{"MOT2":{"ABS_POS":"0.000157220[deg]","COMPENSATION_POS":"-0.000157220[deg]","POSITION":"0.000000000[deg]","STATUS":{"HIZ":1,"UVLO":0,"TH_SD":0,"TH_WRN":0,"OCD":0,"WCMD":0,"NOPCMD":0,"BUSY":0,"DIR":"CCW","MST":"stop"},"HSENDET":1}}}}

GOTO ANGLE
Go to an angle of 20 Degrees
{"req":{"cmd":{"MOT2":{"GOTO":{"DEG":20.0002}}}}}

CALIBRATION
Start calibration: 
{"req":{"set": {"MOT2":{"CAL_STATUS": "exec"}}}}
Response:
{"res":{"set":{"MOT2":{"CAL_STATUS":"done","STATUS":{"HIZ":1,"UV

Asc for motor-calibration status: (still calibrating??)
{"req":{"get": {"MOT2" :{"CAL_STATUS":""}}}}...

Response:
{"res":{"get":{"MOT2":{"CAL_STATUS":"stop","STATUS":{"HIZ":1,"UV...
or
{"res":{"get":{"MOT2":{"CAL_STATUS":"exec","STATUS":{"HIZ":1,"UV...

knro, 2022-02-20:
This looks very similar to the existing driver? perhaps some simple changes are required to make it work with Arco?

ChristianHen, 2022-02-20:
I had the same impression, but unfortunatelly, I don't even know where to begin? Can the rotator driver be integrated into the focuser driver? I suppose so, becouse for the Nightcrawler this seems to be possible, but I have no idea of how to realize that.

ChristianHen, 2022-02-24:
Ok, I managed to realize a rudimentary driver for the Esatto that allows for absolute goto to a given angle. To do so, I have modified the existing sestosenso2 driver. 
I don't know, if it respects any convention, and I am sure, any experienced programmer could have done better. But it moves. If anybody is interested in it, I can share it.

ChristianHen, 2022-02-25:
Could anybody please tell me, what would be the way to get the changes I made to the senstosenso2 driver (Arco-support) into the regular builds?
What are the basic requirements for changes to be accepted? Is there any kind of quality control? What do I have to do?

knro, 2022-02-26:
Can you please share your changes? Did you test other functionalities?

ChristianHen, 2022-02-26:
sestosenso2.cpp

/*
    SestoSenso 2 Focuser
    Copyright (C) 2020 Piotr Zyziuk
    Copyright (C) 2020 Jasem Mutlaq (Added Esatto support)

    This library is free software; you can redistribute it and/or
    modify it under the terms of the GNU Lesser General Public
    License as published by the Free Software Foundation; either
    version 2.1 of the License, or (at your option) any later version.

    This library is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
    Lesser General Public License for more details.

    You should have received a copy of the GNU Lesser General Public
    License along with this library; if not, write to the Free Software
    Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA

*/

#include "sestosenso2.h"

#include "indicom.h"

#include <cmath>
#include <cstring>
#include <memory>
#include <algorithm>

#include <assert.h>
#include <termios.h>
#include <unistd.h>
#include <connectionplugins/connectionserial.h>
#include <sys/ioctl.h>

static std::unique_ptr<SestoSenso2> sesto(new SestoSenso2());

static const char *MOTOR_TAB  = "Motor";
static const char *ENVIRONMENT_TAB  = "Environment";
static const char *ROTATOR_TAB = "Rotator";

struct MotorRates
{
    // Rate values: 1-10
    uint32_t accRate = 0, runSpeed = 0, decRate = 0;
};

struct MotorCurrents
{
    // Current values: 1-10
    uint32_t accCurrent = 0, runCurrent = 0, decCurrent = 0;
    // Hold current: 1-5
    uint32_t holdCurrent = 0;
};

// Settings names for the default motor settings presets
const char *MOTOR_PRESET_NAMES[] = { "light", "medium", "slow" };

SestoSenso2::SestoSenso2() : RotatorInterface(this)
{
    setVersion(0, 7);
    // Can move in Absolute & Relative motions, can AbortFocuser motion.
    FI::SetCapability(FOCUSER_CAN_ABS_MOVE | FOCUSER_HAS_BACKLASH | FOCUSER_CAN_REL_MOVE | FOCUSER_CAN_ABORT);
    //RI::SetCapability(ROTATOR_CAN_ABORT);

    m_MotionProgressTimer.callOnTimeout(std::bind(&SestoSenso2::checkMotionProgressCallback, this));
    m_MotionProgressTimer.setSingleShot(true);
[... 1774 lines cut ...]
    if(targetAngle>getArcoAbsPos()+180.) {targetAngle -= 360.0;}    
     
    snprintf(cmd, sizeof(cmd), "{\"req\":{\"cmd\":{\"MOT2\" :{\"GOTO\":{\"DEG\":%f}}}}}", targetAngle);
    
     //LOG_INFO("Hallo");
     LOGF_INFO("Command sent: %s",cmd);
     LOGF_INFO("Rotator moving to: %f",targetAngle);
    return sendCmd(cmd, "DEG", res);
}

bool CommandSet::isArcoBusy()
{
     char res[SESTO_LEN] = {0};
     
     if(!sendCmd("{\"req\":{\"get\":{\"MOT2\":\"\"}}}", "BUSY", res))
	{
		LOG_INFO("Could not check if Arco is busy.");
		return true;
	}

	int busy = std::stoi(res);
	if(!busy==0) return true;
	
  return false;
}

ChristianHen, 2022-02-26:
sestosenso2.h

/*
    SestoSenso 2 Focuser
    Copyright (C) 2020 Piotr Zyziuk

    This library is free software; you can redistribute it and/or
    modify it under the terms of the GNU Lesser General Public
    License as published by the Free Software Foundation; either
    version 2.1 of the License, or (at your option) any later version.

    This library is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
    Lesser General Public License for more details.

    You should have received a copy of the GNU Lesser General Public
    License along with this library; if not, write to the Free Software
    Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA

*/

#pragma once

#include "indifocuser.h"
#include "inditimer.h"
#include "indirotatorinterface.h"

class CommandSet
{

    public:
        CommandSet(int Port, const char *deviceName)
        {
            PortFD = Port;
            this->deviceName = deviceName;
        }
        int PortFD;
        bool stop();
        bool getSerialNumber(char *res);
        bool getFirmwareVersion(char *res);
        bool abort();
        bool go(uint32_t targetTicks, char *res);
        bool goHome(char *res);
        bool fastMoveOut(char *res);
        bool fastMoveIn(char *res);
        bool getMaxPosition(char *res);
        bool getHallSensor(char *res);
        bool storeAsMaxPosition(char *res);
        bool goOutToFindMaxPos();
        bool storeAsMinPosition();
        bool initCalibration();
        bool getAbsolutePosition(char *res);
        bool getCurrentSpeed(char *res);
        bool applyMotorPreset(const char *name);
        bool applyMotorUserPreset(uint32_t index);
        bool saveMotorUserPreset(uint32_t index, struct MotorRates &mr, struct MotorCurrents &mc);
        bool getMotorTemp(char *res);
        bool getExternalTemp(char *res);
        bool getVoltageIn(char *res);
        bool getMotorSettings(struct MotorRates &ms, struct MotorCurrents &mc, bool &motorHoldActive);
        bool setMotorRates(struct MotorRates &ms);
        bool setMotorCurrents(struct MotorCurrents &mc);
        bool setMotorHold(bool hold);
        bool getARCO(char *res);
	//bool getArcoAbsPos(char *res);
        double getArcoAbsPos();
	bool setArcoAbsPos(double angle, char *res);
	bool isArcoBusy();
        
[... 222 lines cut ...]
         double lastRotatorPosition { 0 };

        typedef enum { BacklashIdle, BacklashMinimum, BacklashMaximum, BacklashComplete } BacklashStage;
        BacklashStage bStage { BacklashIdle };

        //        int m_MotionProgressTimerID {-1};
        //        int m_HallSensorTimerID {-1};
        INDI::Timer m_MotionProgressTimer;
        INDI::Timer m_HallSensorTimer;
        bool m_IsSestoSenso2 { true };
        /////////////////////////////////////////////////////////////////////////////
        /// Static Helper Values
        /////////////////////////////////////////////////////////////////////////////
        // CR is the stop char
        static const char SESTO_STOP_CHAR { 0xD };
        // Update temperature every 10x POLLMS. For 500ms, we would
        // update the temperature one every 5 seconds.
        static constexpr const uint8_t SESTO_TEMPERATURE_FREQ {10};
        // Wait up to a maximum of 3 seconds for serial input
        static constexpr const uint8_t SESTO_TIMEOUT {5};
        // Maximum buffer for sending/receving.
        static constexpr const int SESTO_LEN {1024};
      

};

ChristianHen, 2022-02-28:
Arco support includes now
Goto
Abort
Sync
Cal

see here:
https://github.com/ChristianHen/indi/blob/Add-Esatto-Arco-support/drivers/focuser/sestosenso2.cpp
https://github.com/ChristianHen/indi/blob/Add-Esatto-Arco-support/drivers/focuser/sestosenso2.h

closed 2022-03-24 by knro