indilib/indi #1629: Documentation for Celestron AUX driver: Wiring diagram for AUX cable

nolsum, 2022-02-20:
The developement of the Celestron AUX driver has progressed, but I have problems producing a working AUX cable. This would be with a RJ12, fitting the AUX port on the mount and a 9-pin DSUB fitting a serial-USB-converter. While Pin numbers are given in the Celestron AUX documentation, what is needed is the precise pins on both sides, also the DSUB. That is, a wiring diagram for the RJ12 - DSUB 9-pin cable. 

Various wiring diagrams exist on the internet, however, it is uncler which fits this precise implementation. 

Magnus

T-STUDIO, 2022-03-13:
Celestron's AUX jack doesn't work with a simple UART connection like an EQMOD cable.

You will need an interface to convert the signal.

This circuit is in the hand controller, but it is not sold separately, so you will have to make your own.

https://indilib.org/forum/development/8246-indi-celestron-aux.html?start=228#80785

knro, 2022-06-09:
The driver documentation wiring instructions are not sufficient? if yes, can you propose an update?

T-STUDIO, 2022-06-09:
The description of the AUX part is different.
Since AUX is not a UART and there is no commercial product, circuit design
is required.

T-STUDIO, 2022-06-09:
Checking with mlord on the Cloudy Night Forum may be the fastest.

He is the creator of the AUX-USB converter.

https://www.cloudynights.com/topic/812944-homebrew-mount-usb-connection-for-aux-port/page-5

knro, 2022-07-21:
Thanks, do we have an update for the driver documentation?

nolsum, 2022-07-21:
Hi!

I have not tested the Homebrew device from the Cloudy Night page, as I 
am still unsure whether the aux driver is operational. I will test it in 
the fall (currently rebuilding the pier for my mount). After that, I can 
contribute to the documentation.

Magnus

Den 2022-07-21 kl. 10:57, skrev Jasem Mutlaq:

knro, 2022-07-21:
AUX driver is operation. Ok, will close this issue. Please open if there is a new issue.

closed 2022-07-21 by knro