indilib/indi #1635: Enable nframe support regardless of orientation of device and set default steps/degree

mads0100, 2022-03-05:
Change the phase wires from 2 to 5 in order enable full control.  Change the default steps/degree from 1000 to 55 per dev guidance.

mads0100, 2022-03-05:
Submitted an update to it.

knro, 2022-03-05:
Thanks. Submitted an update where?

mads0100, 2022-03-05:
Github. It’s my first time.

knro, 2022-03-05:
Got a link to the PR? I don't see it.

mads0100, 2022-03-05:
https://github.com/mads0100/indi

I don’t think I did this right.. it’s the nframe.cpp that needed to be updated per Gene. 

Sorry.

knro, 2022-03-06:
Please follow the instructions on how to submit a PR to INDI. Looking forward to the PR!

astrogene1000, 2022-03-09:
Chris, the 'closest direction' nframe test code I sent was test code. To make it 'correct' there needs to be a switch for 'cord wrap' versus 'closest direction'. The original GIT was always cord wrap aware, the test code is always 'closest direction' path.

closed 2024-04-14 by github-actions[bot]