indilib/indi #1640: UART serial connection is not working

T-STUDIO, 2022-03-13:
From Ver1.9.4, UART serial connection is not working with some drivers.

The serial is recognized, but the driver cannot be connected.

The driver I have confirmed this phenomenon is as follows.

・ SkyWatcher-ALTAZ
・ Celestron AUX

knro, 2022-03-13:
What does the log say exactly?

T-STUDIO, 2022-03-14:
Logs cannot be retrieved because the mount cannot be connected.

As I mentioned in the forum, the serial is recognized, but I can no longer connect to the mount.

I want a rollback option as 1.9.3 worked best. .. .. (Returned to 1.9.21 with backup. I also tried 1.9.5 but can't connect.)

T-STUDIO, 2022-06-09:
I have sent logs on the forum several times.

Nekomeshi, 2022-06-13:
I found the latest SkyWatcher-ALTAZ driver doesn't follow the Skywatcher's serial communication protocol.
I compared skywatcherAPI.cpp of master branch and V1.9.2, and it has been updated, especially in TalkwithAxis() that is a  function to communicate with telescope.
I wrote the detail in my blog. I'm afraid but the English is poor because it is translated into English  by Google.

https://nekomeshi312-livedoor-blog.translate.goog/archives/14371265.html?_x_tr_sl=ja&_x_tr_tl=en&_x_tr_hl=ja&_x_tr_pto=wapp

knro, 2022-06-13:
Thank you, can you [enable mount logging in Ekos](https://indilib.org/individuals/logs-howto.html)? Edit your equipment profile and then turn off "Auto Connect". Save your profile and then start it. After it fails to connect to the mount, attach the log file here.

T-STUDIO, 2022-06-14:
SkyWatcher-ALTAZ
https://indilib.org/forum/mounts/11411-skywatcher-virtuso-will-not-be-reconnected-in-ver1-9-4.html

indi_celestron_aux
https://indilib.org/forum/development/8246-indi-celestron-aux.html?start=252#83410

The logs are here.

As Mr.Nekomeshi mentioned, the new driver has a mount check feature that causes some mounts to stop working.

knro, 2022-07-21:
I believe this is already fixed in master, can you test?

Nekomeshi, 2022-07-21:
Yes, it is my pull request.

closed 2022-07-21 by knro