indilib/indi #1645: AZ-GTI Snap option not appearing

killercode, 2022-03-21:
**Describe the bug**
According to this ticket: https://github.com/indilib/indi/issues/1181 this feature was introduced on Indi, but unfortunatly running on stellarmate and astropi I am not able to find the option when choosing the AZ GTI mount.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start service of AZ-GTI Wifi
2. Make a successfull connection, go to options
3. The expected SNAP option doesn't appear

**Expected behavior**
The snap option should appear and shutter control of dlsr cameras

**Desktop (please complete the following information):**
 - OS: stellarmate, astropie

killercode, 2022-03-21:
Some more insights - Apparently the driver selection is wrong.
If I choose the AZ-GTI equatorial it loads the correct driver

[image]

Although for AZ-GTI Wifi ( which is the same device but it should be configured as azimuth ) it is loading other driver that doesn't contain the option

[image]

Can I fix this in runtime without changing the source?

knro, 2022-03-21:
That feature was implement in EQMod driver and not SKyWatcher Alt-Az driver this is why you see it in the former but not the latter.

knro, 2022-03-22:
Right, but unless a developer contributes this feature to the SkyWatcher Alt-Az driver, it's not going to be available.

killercode, 2022-03-23:
Yeah I could do it but I am unfortunatly out of time... 

Seems like it's not a big thing to implement, as I saw the implementation on the Indi-3rdparty implementation for the eqmount and seems fair simple, although I saw the structure of the AltAzMount and seems a completly different structure so it would take a big efford to check the code and understand where to change.

Let's hope in the following days a developer jumps on this, otherwise I will try to pick it up!

closed 2022-07-21 by knro