indilib/indi #1647: Skywatcher Alt Az Simple new issue ?

lconti73, 2022-03-25:
### Discussed in https://github.com/indilib/indi/discussions/1614

<div type='discussions-op-text'>

<sup>Originally posted by **saviourbot** February  3, 2022</sup>
After an upgrade my Alt Az will no longer connect (Debug below) I dont want to raise it as an issue if it is just something I am doing wrong has there recently been a change in InitMount() ?

The mount connests as a USB device to the PI:
`astroberry@astroberry:~ $ ls -l /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0
lrwxrwxrwx 1 root root 13 Feb  3 10:26 /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0 -> ../../ttyUSB0
`

Debug output shows no connect - Handshake failed:
INFO	0.030534 sec	: Session log file /home/astroberry/.indi/logs/2022-02-03/indi_skywatcherAltAzMount/indi_skywatcherAltAzMount_10:51:52.log
DEBUG	0.121657 sec	: Connecting to /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BKCXb106315-if00-port0 @ 9600
DEBUG	0.126972 sec	: Port FD 3
DEBUG	0.127087 sec	: Connection successful, attempting handshake...
SCOPE	0.127124 sec	: SkywatcherAPIMount::Handshake
SCOPE	0.127159 sec	: CheckIfDCMotor
SCOPE	1.129368 sec	: CMD <e1>
SCOPE	1.133705 sec	: RES <e1>
SCOPE	1.133787 sec	: Mount type not supported. 0
SCOPE	1.133817 sec	: SkywatcherAPIMount::Handshake - Result: 0
DEBUG	1.133846 sec	: Handshake failed.

`bool SkywatcherAltAzSimple::Handshake()
{
    DEBUG(DBG_SCOPE, "SkywatcherAltAzSimple::Handshake");
    SetSerialPort(PortFD);

    Connection::Interface *activeConnection = getActiveConnection();
    if (!activeConnection->name().compare("CONNECTION_TCP"))
    {
        tty_set_generic_udp_format(1);
    }

    bool Result = InitMount();

    if (getActiveConnection() == serialConnection)
    {
        SerialPortName = serialConnection->port();
    }
    else
    {
        SerialPortName = "";
    }

    RecoverAfterReconnection = false;
    DEBUGF(DBG_SCOPE, "SkywatcherAltAzSimple::Handshake - Result: %d", Result);
    return Result;
}`

It has been working fantastic but since an apt upgrade not so much. 

Thanks in advance
Gus

Update: I did a fresh build of the astroberry 2.0.4 build and it worked. Then I did an apt update && apt upgrade and I got the error again.

So, it appears to be a recent change causing the problem.</div>

knro, 2022-03-26:
I don't think this driver was updated recently. Why are you using this driver and not the regular SkyWatcherAltAz ?

lconti73, 2022-03-26:
hello, 

The SkyWatcherAltAz driver doesn't work for me either.

In the previous version it did work.

I am using astroberry 2.0.4, and after doing update & upgrade & dist-upgrade the Skywatcher AltAz drivers have stopped working.

Sorry for forwarding another user's issue. But the same issue happens to me. It's the same "Handshake failed" error after connecting to the pl2303 interface successfully.

thanks in advance

T-STUDIO, 2022-06-09:
I am no longer connected from 1.9.4.

I have reported several times in the forum.

knro, 2022-07-21:
SkyWatcher Alt-Az simple is now removed, please only use SkyWatcher Alt-Az driver.

closed 2022-07-21 by knro