indilib/indi #1649: FILTER keyword not written in Flat files - randomly

wjdrijfhout, 2022-03-31:
Not all Flat files written by Ekos have the FILTER keyword added to the FITS header. Only occasionally the keyword is written in the file. In a recent batch of 25 Flat frames for LRGB filters (100 total), the keyword was only written in file 9 and 15 of Green, and file 1 of Red. When adding the flat files to the WBPP script in PixInsight, it does not know which flats files to correct which Light files. 

The issue seems to have started with version 3.5.7. In [this thread](https://indilib.org/forum/focusers-filter-wheels/11524-filter-name-in-fits-header-intermittently.html), the exact same issue is described for two quite different setups, with different filter wheels, different operating systems and different server/client architecture.

All Flat files should have the FILTER keyword written in the FITS header with the proper filter setting

OS: Both RPi and Mac, supposedly since 3.5.7

knro, 2022-03-31:
Any way to reproduce this behavior?

wjdrijfhout, 2022-03-31:
Shot 175 flats on two sessions (March 10 and March 19). 3 out of 175 frames had the FILTER keyword added.

knro, 2022-03-31:
That's not exactly a step-by-step procedure to reproduce the issue. What equipment exactly did you have in your profile?

wjdrijfhout, 2022-03-31:
Ah, sorry. This was the profile:
[image]
But keep in mind that exactly the same issue has been reported by someone who used a very different set of equipment:
https://indilib.org/forum/focusers-filter-wheels/11524-filter-name-in-fits-header-intermittently.html

alepensato, 2022-08-10:
The problem is present also in my two setup with the QHYCFW3, and not only with FLAT, but also with LIGHT. In these days I captured 107subs usinge the SII filter in in just 3 of each file i have the tag with the filter. In the other setup about of the 60% have the filter flag. This problem is present since the KStars 3.5.8. As reported from many user this issue is present with different setup and it is random

wberlo, 2022-08-30:
The problem occurs for both Lights and Flats on a Raspberry Pi 4 with an ASI294MM, ASI efw, Pagasus Focuscube, and EQMod mount (SkyWatcher AZ-EQ6). Kstars v 3.6.0, fresh install. The problem is random, but in general more than half of the exposures are affected.

Regards,

Wim

Ruegs, 2022-09-27:
Over the last two weeks I've run into this issue of the filter not being written to the FITS header. Since I've had no changes to my environment I had assumed I may have inadvertently flipped a software switch somewhere along the way but I've no idea what that might have been. I've just finished importing a few hundred images I've taken with EKOS/KStars (3.6.0 Stable running on MacOS 12.6 (Monterey)) remotely controlling an RPi4 at the mount and there is no filter record in the FITS header. 60+ other records but no filter information. This is happening for all my Light and Flat frames of all kinds of exposure lengths taken over the last 10 days - no exceptions. I've been using EKOS/KStars for over a year and haven't run into this before. I am happy to provide any information that can provide insights.

knro, 2022-11-22:
This should be fixed by KStars v3.6.2, not an INDI issue.

closed 2022-11-22 by knro