indilib/indi #1659: 360 after restarting the mount from unparked position

M31Fan, 2022-05-07:
**Describe the bug**
This issue happens with EKOS, but I haven't been able to verify with other software
Every time that I don't park my mount, and then cut the power, once I reconnect it for the first time (i.e. after a power outage), and try to move the mount, it will always do a full turn (Eastward) on the RA axle. This only happens once and always on the first time I try to move after a power-off without parking.
If I park, OR if the power is kept, once I reconnect the mount and try to move, it behaves normally. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. Turn system on, 
2. connect and move your mount a bit away from the NCP,
3. Power off without parking,
4. Re-power, connect,  and try to move. 
5. Careful with the cables, as the full turn can mess with them. 

**Expected behavior**
Full 360 turn, once. After that is normal. 

**Screenshots**
N/A

**Desktop (please complete the following information):**
- Astroberry, multiple versions, included the latests

**Log Files**
No log available at this time.

TallFurryMan, 2022-05-07:
Which driver are you using? Most of the drivers consider the initial position of the mount is the parking position for safety purposes. Then, the question is to determine whether it is the driver that requests the turn, or the firmware when receiving the unpark request.

knro, 2022-11-22:
This is *most likely* not related to INDI driver but probably a behavior of the mount itself on power off?

closed 2022-11-22 by knro