indilib/indi #1662: indi_pegasus_upb crashes if project compiled with _GLIBCXX_ASSERTIONS

wgauvin, 2022-05-28:
**Describe the bug**
When the project is compiled with `_GLIBCXX_ASSERTIONS` in the `CXXFLAGS` then `indi_pegasus_upb` crashes during the getSensorData step. When in debug logs it is clear that the `getSensorData("PA")` does the communication with my UPBv2 as response comes back. However, the driver doesn't complete and the program aborts() (sig = 6).

The occurs in ArchLinux distros (main ArchLinux as well as Manjaro ARM) as the default CXXFLAGS for packages includes `_GLIBCXX_ASSERTIONS`. I have done local builds removing that flag and the driver doesn't crash.

When the `_GLIBCXX_DEBUG` is also included, the details state that there is an assertion of trying to index a position 0 when the size of the vector is 0.

I've tracked this down to 2 parts within the `pegasus_ubp.cpp`, in `getSensorData` and `sensorUpdated` there is no check to see if the `lastSensorData` has been set or not. Also this issue also exists in `getStepperData` (found by first doing a local fix for `sensorUpdated`.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Build project with CXXFLAGS including `_GLIBCXX_ASSERTIONS` set (also include `_GLIBCXX_DEBUG` for details)
2. run `./indiserver -vvv ./indi_pegasus_upb`
3. Enable INDI debugging
```
indi_setprop "Pegasus UPB.DEBUG.ENABLE=On"
indi_setprop "Pegasus UPB.DEBUG_LEVEL.DBG_DEBUG=On"
indi_setprop "Pegasus UPB.LOGGING_LEVEL.LOG_DEBUG=On"
indi_setprop "Pegasus UPB.LOG_OUTPUT.FILE_DEBUG=On"
```
5. Try to have INDI server to connect to the UPBv2
```
indi_setprop "Pegasus UPB.CONNECTION.CONNECT=On"
```

**Expected behavior**
The expected behavior is that `indiserver` will connect to the device. However, when compiled with `_GLIBCXX_ASSERTIONS` the program aborts.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: ArchLinux
 - Version: Linux deepthought 5.17.9-arch1-1 #1 SMP PREEMPT Wed, 18 May 2022 17:30:11 +0000 x86_64 GNU/Linux 

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[indi_pegasus_upb_fixed.log](https://github.com/indilib/indi/files/8790653/indi_pegasus_upb_fixed.log)
[indi_pegasus_upb_invalid.log](https://github.com/indilib/indi/files/8790654/indi_pegasus_upb_invalid.log)

In the invalid log, on line 335 you can see the assertion has failed
In the valid log, line 427 is the response of the "PA" call and what follows is that it processes correctly

wgauvin, 2022-05-28:
I am happy to do a pull request to fix this as I have a hack to work around it.

knro, 2022-05-28:
Thank you for the report. Sure please send a PR if ready.

wgauvin, 2022-05-28:
PR is #1663

commit 37860f05 (2022-05-28): Ensure defensive check of UBP previous data to fix crash - fixes #1662 (#1663)
closed 2022-05-28 by knro