indilib/indi #1678: [CCD Drivers] Expose details of native encoding

pludov, 2022-06-25:
Following https://github.com/indilib/indi/pull/1630, on CCD drivers, the new ENCODE setting can be set to Native, so FITS is bypassed and raw data from the driver are sent as-is. This is very interesting for performance constrained clients !

However, there seem to be no reliable way to determine the exact format of the data from a client POV.

I suggest adding a new property in "Image Info" tab, that exposes the information for the current CCD_CAPTURE_FORMAT (suggest CCD_CAPTURE_FORMAT):
  * endianness
  * naxes (or RGB ? or sample per pixel ?)

Also I think default endianness should depend on the system (sending uint16_t will result in different endianesses on different CPU)

pludov, 2024-04-07:
This is still relevant. The endianness is also a performence subject, to avoid convertions where possible

knro, 2024-04-07:
Ok understood. How do you propose we do that?

closed 2024-06-15 by github-actions[bot]