indilib/indi #1681: iOptronV3 - Track mode automatically changes to Lunar

evripidis, 2022-06-26:
**Describe the bug**
"indi_ioptronv3_telescope"

When you are connected to mount's internal wifi hotspot, the track mode automatically changes from Sidereal to Lunar. This doesn't occur when you are connected through hand controller.

Indilib thread: https://www.indilib.org/forum/ekos/11170-track-mode-automatically-changes-to-lunar.html

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect computer's wifi to mount's wifi hotspot 
2. Start Kstars/Ekos and run the profile with the driver indi_ioptronv3_telescope as driver for the GEM28
3. Start session and after a few moments the a Timeout Error appears on the console and the track mode changes to Lunar.

**Expected behavior**
Not changing the track mode.

knro, 2022-06-27:
Need mount logs to see what triggers this. Please attach the logs.

evripidis, 2022-06-28:
Hello @knro ,

I'm gonna recheck it from scratch cause I just gave up 4 months ago when I started the thread on indilib. My bad, I should have reported directly here.

As far as I can understand the pasted text there is not any useful correct?

`[2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - Welcome to KStars 3.5.7 Stable [2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - Build: 2022-01-18T17:14:21Z [2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - OS: "ubuntu" [2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64" [2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - Arch: "x86_64" [2022-01-25T21:37:22.571 EET INFO ][                org.kde.kstars] - Kernel Type: "linux" [2022-01-25T21:37:22.572 EET INFO ][                org.kde.kstars] - Kernel Version: "5.13.0-27-generic" [2022-01-25T21:37:22.572 EET INFO ][                org.kde.kstars] - Qt Version: 5.12.8 [2022-01-25T21:37:23.059 EET DEBG ][                org.kde.kstars] - Opened the User DB. Ready. [2022-01-25T21:37:23.321 EET INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3 [2022-01-25T21:37:23.321 EET INFO ][                org.kde.kstars] -   Sky Mesh Size:  512 [2022-01-25T21:37:23.359 EET INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat" [2022-01-25T21:37:23.361 EET INFO ][                org.kde.kstars] - Processing  "deepstars.dat" , HTMesh Level 3 [2022-01-25T21:37:23.361 EET INFO ][                org.kde.kstars] -   Sky Mesh Size:  512 [2022-01-25T21:37:23.361 EET INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "deepstars.dat" [2022-01-25T21:37:23.949 EET WARN ][                org.kde.kstars] - "Star HD20.794 not found." [2022-01-25T21:37:23.974 EET WARN ][                org.kde.kstars] - "Star HD98.230 not found." [2022-01-25T21:37:24.022 EET INFO ][                org.kde.kstars] - Loaded DSO catalogs. [2022-01-25T21:37:24.287 EET INFO ][                org.kde.kstars] - Loading comets [2022-01-25T21:37:24.565 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/16/" [2022-01-25T21:37:24.565 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/22/" [2022-01-25T21:37:24.565 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/24/" [2022-01-25T21:37:24.565 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/32/" [2022-01-25T21:37:24.566 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/48/" [2022-01-25T21:37:24.566 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/64/" [2022-01-25T21:37:24.566 EET WARN ][                       default] - Invalid Context= "stock" line for icon theme:  "/usr/share/icons/ubuntu-mono-dark/stock/128/" [2022-01-25T21:37:24.572 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/8x8/legacy/" [2022-01-25T21:37:24.572 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/16x16/legacy/" [2022-01-25T21:37:24.572 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/16x16/ui/" [2022-01-25T21:37:24.572 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/22x22/legacy/" [2022-01-25T21:37:24.573 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/24x24/legacy/" [2022-01-25T21:37:24.573 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/24x24/ui/" [2022-01-25T21:37:24.573 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/32x32/legacy/" [2022-01-25T21:37:24.573 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/32x32/ui/" [2022-01-25T21:37:24.573 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/48x48/legacy/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/48x48/ui/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/64x64/legacy/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/64x64/ui/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - Invalid Context= "Legacy" line for icon theme:  "/usr/share/icons/Adwaita/96x96/legacy/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - Invalid Context= "UI" line for icon theme:  "/usr/share/icons/Adwaita/96x96/ui/" [2022-01-25T21:37:24.574 EET WARN ][                       default] - [... cut ...]           org.kde.kstars.ekos] - Resetting Ekos Manager... [2022-01-25T21:44:39.614 EET WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work. [2022-01-25T21:44:39.614 EET WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed. [2022-01-25T21:44:39.691 EET INFO ][           org.kde.kstars.ekos] - "INDI services stopped."`

evripidis, 2022-06-28:
@knro I've just tried it and after just slew to a random area the issue came up . I hope this file to be ok. 
[log_20-26-41.txt](https://github.com/indilib/indi/files/9003452/log_20-26-41.txt)

sonny486, 2022-06-28:
Hmm the log shows a disconnect, then an indi failure, similar to what I am seeing when trying to capture on my ZWO camera, the camera detaches, and reattaches.  After several attempts, the INDI server crashes and closes all connections.

https://github.com/indilib/indi-3rdparty/issues/592

evripidis, 2022-07-06:
> I can't read it, it needs to be attached as a file. Mount logging must be enabled before logs are collected for that.

@knro When you have time, please just confirm that is the correct log that you were looking for.

evripidis, 2022-07-20:
@knro Any idea for this please?

knro, 2022-07-21:
No debug tracking in there. You checked both INDI + Mount (under drivers)? If yes, go to iOptron --> Options and make sure all the debug checkboxes are on. Because the log does not show the verbose logs we need to diagnose this.

evripidis, 2022-07-23:
@knro Finally I've managed to install indi & Kstars on a fresh unit and I can replicate the issue instantly. Here is the correct full log I think this time.

[log_00-53-31.txt](https://github.com/indilib/indi/files/9174731/log_00-53-31.txt)

knro, 2022-07-24:
@evripidis Thank you that was helpful. Please try to update from latest INDI from Git and report back with the log (even if it works, upload the logs).

evripidis, 2022-07-29:
> @evripidis Thank you that was helpful. Please try to update from latest INDI from Git and report back with the log (even if it works, upload the logs).

@knro I do not know if this has to do with the wifi option, but after installing the latest version, (and trying to troubleshoot the Toupcam issue) my mount crashes. Logs here
[log_23-42-14.txt](https://github.com/indilib/indi/files/9223203/log_23-42-14.txt)
[image]
.

knro, 2022-07-29:
After timeout apparently the mount starts sending the previous command response.. I added a workaround. Can you test? it needs to be compile from libindi master. If you confirm it fixes the issue, I'll publish an update to the package.

evripidis, 2022-07-30:
> After timeout apparently the mount starts sending the previous command response.. I added a workaround. Can you test? it needs to be compile from libindi master. If you confirm it fixes the issue, I'll publish an update to the package.

@knro Ok I could say I'm officially lost.. Tried to compile INDI, tried to install individually the required drivers and libraries and now, I can't load my profile cause says "Ekos requires at least on CCD or guider to operate".

I've tried to create a new profile and the toupcam drivers are not there... :-/ I see them libtoupcam into my Project folder but Ekos is not :-(

evripidis, 2022-07-30:
After many hours messing around and trying to fix duplicates, overrides, discontinuities etc. I had no luck... Thank fully I had TimeShfit on so it was quite easy to revert back.

@knro It would be great if you pass tho changed to the nightly so I could continue there the troubleshooting. Just give me a heads up. Until then, back to the cable :-(

knro, 2022-07-30:
Is there a chance you can setup teamviewer? this would be much quicker.

evripidis, 2022-07-30:
> Is there a chance you can setup teamviewer? this would be much quicker.

Of course, no problem, any time!

knro, 2022-07-30:
ok please send email to support@stellarmate.com with the login details (ID + Password) and make sure the mount is connected.

evripidis, 2022-07-30:
> ok please send email to [support@stellarmate.com](mailto:support@stellarmate.com) with the login details (ID + Password) and make sure the mount is connected.

Consider it done

evripidis, 2022-08-02:
@knro Please take a look on this thread: https://www.indilib.org/forum/ekos/10122-meridian-flip-not-triggering.html?start=12#84944

Is it possible those changes above to trigger this issue in MF ?

evripidis, 2022-08-03:
@knro Something is going on with the driver. I've created a new profile, only with the mount (connected with usb & HC) and CCD simulator and after a few minutes tracking I can see errors again and goes to idle.

[log_12-09-22.txt](https://github.com/indilib/indi/files/9250104/log_12-09-22.txt)

knro, 2022-08-03:
I see it. No commands given. It's either power issue or something wrong in the mount electronics/hardware. This is not a software issue.

evripidis, 2022-08-03:
> I see it. No commands given. It's either power issue or something wrong in the mount electronics/hardware. This is not a software issue.

Please ignore the previous log, it is not from a fresh profile. This is the correct one.

[log_12-29-06.txt](https://github.com/indilib/indi/files/9250339/log_12-29-06.txt)

knro, 2022-08-04:
No differences.

evripidis, 2022-08-20:
It seems that the latest changes resolved the issue

evripidis, 2022-10-15:
Hello @knro 

Any idea why is that back ?

[log_21-02-01.txt](https://github.com/indilib/indi/files/9791731/log_21-02-01.txt)

[image]

knro, 2022-10-15:
From the logs:

```
[2022-10-07T21:36:03.885 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] RES <+3240000004370577021> "
[2022-10-07T21:36:04.470 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] CMD <:MH#> "
[2022-10-07T21:36:04.470 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] RES <+> "
[2022-10-07T21:36:04.886 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] CMD <:GLS#> "
[2022-10-07T21:36:04.886 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] RES <0796500047022000010911> "
[2022-10-07T21:36:04.890 EEST INFO ][           org.kde.kstars.indi] - iOptron GEM28 :  "[ERROR] bool IOPv3::Driver::getStatus(IOPv3::IOPInfo*): Expected 23 bytes but received 22. "
[2022-10-07T21:36:04.890 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] CMD <:GEP#> "
[2022-10-07T21:36:04.890 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] RES <+3240000004370572521> "
[2022-10-07T21:36:05.888 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] CMD <:GLS#> "
[2022-10-07T21:36:05.888 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] RES <1+0796500047022000020911> "
[2022-10-07T21:36:05.894 EEST INFO ][           org.kde.kstars.indi] - iOptron GEM28 :  "[ERROR] bool IOPv3::Driver::getStatus(IOPv3::IOPInfo*): Expected 23 bytes but received 24. "
[2022-10-07T21:36:05.895 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] CMD <:GEP#> "
```

You see it is issued Go home :MH and instead of responding by "1" as expected, we get "+" which is part of the :GLS response. and then the "1" is receipved later on. Maybe you can increase driver polling from 1000 to 2000 and see if this improves this.

evripidis, 2022-10-16:
@knro It seemed to be promising at the begging but then I had this 
(while actually the session was keep going).

`[2022-10-16T22:44:48.795 EEST DEBG ][     org.kde.kstars.ekos.guide] - PHD2: request: "{\"id\":2490,\"jsonrpc\":\"2.0\",\"method\":\"get_lock_position\"}"
[2022-10-16T22:44:48.828 EEST DEBG ][     org.kde.kstars.ekos.guide] - PHD2: response: "{\"jsonrpc\":\"2.0\",\"result\":[347.65,588.10],\"id\":2490}\r\n"
[2022-10-16T22:44:49.297 EEST DEBG ][     org.kde.kstars.ekos.mount] - **Mount status changed from  "Tracking"  to  "Error"**
[2022-10-16T22:44:49.562 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] CMD <:GLS#> "
[2022-10-16T22:44:49.562 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] RES <+1586568211268141001> "
[2022-10-16T22:44:49.563 EEST INFO ][           org.kde.kstars.indi] - iOptron GEM28 :  "[ERROR] bool IOPv3::Driver::getStatus(IOPv3::IOPInfo*): Expected 23 bytes but received 20. "
[2022-10-16T22:44:49.563 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] CMD <:GEP#> "
[2022-10-16T22:44:49.615 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[SCOPE] RES <+0796500047022000010811> "
[2022-10-16T22:44:49.616 EEST INFO ][           org.kde.kstars.indi] - iOptron GEM28 :  "[ERROR] bool IOPv3::Driver::getCoords(double*, double*, IOPv3::IOP_PIER_STATE*, IOPv3::IOP_CW_STATE*): Expected 20 bytes but received 23. "
[2022-10-16T22:44:50.121 EEST DEBG ][           org.kde.kstars.indi] - iOptron GEM28 : "[DEBUG] CMD <:ZQ00026#> "`

[log_20-39-41.txt](https://github.com/indilib/indi/files/9795968/log_20-39-41.txt)

knro, 2023-05-30:
I created a [PR 1093](https://github.com/indilib/indi/pull/1903) that needs to be tested. Please report back.

evripidis, 2023-05-31:
@knro I didn't have much time, I've tried only with the hand controller connected, with no issue just like the last time. So I can verify that the issue appears only when the mount is connected via wifi.

Since the last time, I've changed my mini PC, so there is a different wifi chip. 

Probably something has to do with mount's wifi ? With the code? At least there are no disconnections.

evripidis, 2023-05-31:
No, wifi works fine. The mini PC (stellarmate) automatically connects to the mount's wifi hotspot with no dropped connections. I'm sure there is nothing wrong with mini PC's wifi chip, cause I've already tested it with two different models.

A few months ago I saw that I have no errors if I connect my mount to the mini PC via the Hand controller and USB cable.
While I connect it via the wifi, intermittently I got those errors.

So from my point of view - I am far more than an expert:

1. There is something with the mount's wifi chip, even if there are no dropped connections (poor quality, slow response on commands etc.) 
2. Something in the code - don't have a clue on that.

(Yes, I didn't try this PR version yet.)

knro, 2023-06-01:
Thanks, let me know when you have time to test the PR yet since I need confirmation is works OK without any regressions.

JulesArcher, 2023-09-26:
Hi @knro I have also commented on #1903, but wanted to include @evripidis in the reply. As I mentioned, I am getting the same reported errors:
[2023-09-25T18:01:01.845 PDT INFO ][ org.kde.kstars.indi] - iOptron CEM26 : "[ERROR] Error setting RA/DEC. "
[2023-09-25T18:01:46.918 PDT INFO ][ org.kde.kstars.indi] - iOptron CEM26 : "[ERROR] bool IOPv3::Driver::getCoords(double*, double*, IOPv3::IOP_PIER_STATE*, IOPv3::IOP_CW_STATE*): Expected 20 bytes but received 23. "

I am no SW expert, so I would need a new  indi_ioptronv3_telescope driver file with the fix to be able to test  it on my system. Could you please send it to me? If so I will gladly test.

Thanks!

evripidis, 2024-01-14:
> Hi @knro I have also commented on #1903, but wanted to include @evripidis in the reply. As I mentioned, I am getting the same reported errors: [2023-09-25T18:01:01.845 PDT INFO ][ [...]

I was connecting my mount to Stellarmate's wifi hotspot. 

Now after this thread https://indilib.org/forum/ekos/14026-has-anyone-controlled-an-ioptron-mount-over-wifi-from-ekos.html I had the idea to connect my mount directly to my home wifi. That it was a success but the issue remains. 

[image]

closed 2024-04-14 by github-actions[bot]