indilib/indi #1682: Fix INDI Docker build

knro, 2022-06-27:
Since libev-ev was added, all builds for Linux distributions are failing. How do we trigger docker rebuild?

Would appreciate any help on resolving this, and perhaps how to fix it in case it happens again in the future.

@abecadel
@jochym 
@eric-vickery

abecadel, 2022-06-27:
@knro  you need to manually trigger this action https://github.com/indilib/indi/actions/workflows/docker.yml - there's a button to do it

knro, 2022-06-30:
Thanks!! I can't believe it was this simple. Now need to figure out why Fedora is failing along with MacOS.

knro, 2022-07-03:
Any idea on what is causing the Fedora failure? Usually @xsnrg would be able to help in this but I think he is busy now.

pawel-soja, 2022-10-17:
@knro I think the issue can be closed.

closed 2022-10-17 by knro