indilib/indi #1701: INDIPrefix is no longer being sent from INDI Driver on MacOS to child driver processes.

rlancaste, 2022-07-15:
**Describe the bug**
This issue was reported in the INDI forum.  It might be a MacOS issue, but might not be.  I have been investigating it since I am usually the one who fixes Mac issues, but this one is a bit tricky.  On MacOS, when KStars launches the INDIServer as a child process running locally, The INDIPrefix variable is set as an environment variable.  INDIServer then launches a driver and is supposed to pass on that environment variable to any child processes like various drivers.  AAG Cloudwatcher needs this variable in order to run buildSkeleton so it can find the skeleton file.  As of June 17th, this worked fine, but after the fast blob changes the next day, a lot of INDIServer code got changed and now that environment variable is not being passed on to the Cloudwatcher driver and probably no other drivers that might need it.  On MacOS, the INDIPrefix is actually very important since INDI including all the drivers, driver support files, and other needed items, is contained in KStars and can be moved around by the user at will.

In my testing, I can confirm that in fact the prefix environment variable is properly set by kstars, and then it gets into the LocalDvrInfo::start() method and still has the variable.  But when it gets to line 1204 or so, it encountered this:

if (fifo)
                unsetenv("INDIPREFIX");

And loses the prefix before the driver is started on line 1215: execlp(name.c_str(), name.c_str(), NULL);

This line is very very similar to the line of code that was there before, which looks like this:

if (fifo.fd > 0)
                unsetenv("INDIPREFIX");

But for some reason, it didn't lose the prefix in the past but it does now.

I haven't checked yet why yet, but there must be something going on with the fifo file.  I can test that later today or tomorrow.

**To Reproduce**
Exact steps to reproduce the behavior.
1.  Run KStars on MacOS. 
2.  Run Device Manager in the menu.
3.  Select the AAG Cloudwatcher driver and click start4. 
4.  It doesn't find the skeleton file, it looks in the share folder of where the driver was built, which might not even exist on the user's computer if it was built on another computer and distributed by craft etc. (that is the Linux location for the file)

**Expected behavior**
A clear and concise description of what you expected to happen.
AAG Cloudwatcher should get the correct skeleton file location which is in KStars.app/Contents/Resources/DriverSupport/
This still worked perfectly a few weeks ago, but now fails.

**Desktop (please complete the following information):**
 - OS: MacOS

Please reference this forum posting
https://indilib.org/forum/ekos/11946-aag-cloudwatcher-broken-again-in-latest-nightly.html

knro, 2022-07-15:
@pludov Can you please check?

rlancaste, 2022-07-15:
Yes I just added an if statement in that function to print whether fifo is true in the new indiserver and to print whether fifo.fd > 0 in the old one before the fast blob changes.  When started by KStars in the new version, fifo is true if INDIServer is started by KStars.  When running the older version of INDIServer and started by KStars, fifo.fd > 0 is False.  So that is definitely different.

rlancaste, 2022-07-16:
One additional observation.  If you start INDI from the command line, it works both before and after the fast blob change as long as you set INDIPrefix as an environment variable.  But when started from KStars, then it unsets the INDIPrefix environment variable and that prevents the driver from working properly.  For some reason in that case, fifo seems to be true.

rlancaste, 2022-07-16:
I looked a little more, it looks like in the new version of the code, the fifo file is created during the processing of the arguments when the INDIServer is starting up.  In the old version of the code, it looks like that happens after the start driver method has run.  This could explain the if statement is now returning true but it was false before.  Thoughts?

pludov, 2022-07-16:
@Jasem Mutlaq ***@***.***> : why is the code clearing
indi_prefix ? Could we just drop the clearing code and always keep
INDI_PREFIX  ?

knro, 2022-07-16:
No we have to clear the environment variable otherwise it might inadvertently affect other drivers.

rlancaste, 2022-07-16:
I don't understand that?  If INDIPrefix has not been set, then not clearing it won't have any effect.  If it has been set, then the drivers need that in order to find files like skeleton files etc.  Why are we clearing it?  

Also It seems like the code before the fast blob changes started the drivers with INDIPrefix still set, but then made the fifo later, so the code to check whether fifo was made and then unset INDIPrefix would have never run because the fifo was not created till later.

pludov, 2022-07-17:
I guess clearing is here so that FIFO orders will be interpreted the same, whatever the env of the server is.

Anyway, I pushed an untested restoration of that behavior here: https://github.com/indilib/indi/pull/1702
Please test and report !

rlancaste, 2022-07-17:
Ok I will be able to test that when I get home later today.  In fact I think the change went in prior to the time craft builds, so it should be fixed in the craft build today which makes it easy to test

knro, 2022-07-19:
Can you debug the server to see where does this happens exactly?

rlancaste, 2022-07-21:
The indiprefix is cleared in the main method on line 1032 of indiserver.cpp: unsetenv("INDIPREFIX");
The prefix is needed by the cloudwatcher driver when it tries to find the skeleton file to run the buildskeleton method on line 217 in basedevice.cpp.

I ran a test and I can confirm that in fact the main method gets all the way down to the line 1037 of indiserver.cpp that says "loop.loop();" before the buildSkeleton method is even run by the driver, which means that the prefix is cleared before that code can be run and it won't get the proper prefix for the skeleton file.  My hope was that by moving the line that unsets the environment variable down after the start command was issued, that it would have the prefix when it was starting up, but since it is running asynchronously, I guess it doesn't get a chance to use the prefix before it is unset.

knro, 2022-07-22:
I think I am starting to see what's going on. The assumption was made that INDIPREFIX is used for each driver just like INDISKEL , INDIDEV, and INDICONFIG, but it's being set per server, not per driver.

So two ways to resolve this:
1. Add INDIPREFIX officially as another FIFO parameter (it's not now) and then make clients define it per driver.
2. Do not unset INDIPREFIX at all.

@pludov @rlancaste Thoughts?

pludov, 2022-07-22:
Prefix is already available from fifo :
https://github.com/indilib/indi/blob/348abc96c1d533bbb139db8775bea99f10f39cda/indiserver.cpp#L1727

rlancaste, 2022-07-22:
Note, I think the whole problem is actually in the core INDI code, not in the 3rd Party drivers. The 3rd Party driver isn't really making use of the variables at all, it just passes the file name.  The main method that unsets the prefix environment variable is in the indiserver code and also the buildSkeleton method that is called by the driver which then assumes that the environment variable still exists is also in indi code, in this case it is part of the indidriver library.

rlancaste, 2022-07-22:
The driver never actually used the indiprefix at all.

rlancaste, 2022-07-24:
Ok just an update, after digging and digging and working through things,  We finally came to the conclusion that it would just be easiest to get rid of that unset on MacOS.  Now it should be fixed.  I will ask the user to test it and see.

rlancaste, 2022-07-24:
https://github.com/indilib/indi/commit/0668feba7623c961e58edff62856dfb16e7b66b8

closed 2022-07-24 by knro