indilib/indi #1703: FreeBSD: indi_v4l2_ccd: capture not working

andkem, 2022-07-19:
**Describe the bug**
When attempting to use PHD2 for guiding using a webcam connected via indi_v4l2_ccd on FreeBSD does not work on latest master. A git bisect tracked the issue as being introduced with #1630. After said merge, images won't be captured and "Invalid source buffer size, skipping frame..." is output in the log and it is impossible to stop the capture without killing PHD2. See attached debug log.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run indiserver -v -l /tmp/ indi_celestron_gps indi_v4l2_ccd
2. Run PHD2
3. Connect to Indi
4. Start capture
5. See that no images are captured

**Expected behavior**
I expect to see images captured from the webcam.

**Desktop (please complete the following information):**
 - OS: FreeBSD 12.3-RELEASE-p5
 - Versions >= git revision 738b078c652c7d9e469075d5d72a96ec4cc75274

**Log Files**
[indi_v4l2_ccd_22:40:32.log](https://github.com/indilib/indi/files/9145145/indi_v4l2_ccd_22.40.32.log)

knro, 2022-07-20:
Unclear how that would ONLY affect FreeBSD?

andkem, 2022-07-20:
Apologies if I've been unclear. I've seen the issue on FreeBSD and don't have a Linux machine to test on at the moment. It is possible the issue exists on Linux as well. I can remove FreeBSD from the bug report title if you want.

andkem, 2022-07-20:
That fixes the problem!

PHD2 in itself should support colour, but I've never tried it with my webcam and have always left it on grayscale. Changing the setting to colour for the working version made it stop working. I can't find much at all in the PHD2 documentation regarding V4L2 and Indi so I don't know if it is supposed to work. I've tried playing video from my camera with a video player and colour works there.

After the commit 738b078c652c7d9e469075d5d72a96ec4cc75274, it would seem my config was switched to colour without me doing anything. The option was renamed so maybe colour was the new default? 

Since this is most likely a PHD2 problem and not an Indi one, I'll close this issue.

Thank you for your help!

closed 2022-07-20 by andkem