indilib/indi #1705: Unable to load Toupcam drivers

evripidis, 2022-07-23:
**Unable to load Toupcam drivers**
indi_toupcam_ccd unable to load and crashes.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. See error

**Screenshots**
[image]
)

**Desktop (please complete the following information):**
 - OS: Lubuntu

**Log Files**
[log_00-24-50.txt](https://github.com/indilib/indi/files/9174744/log_00-24-50.txt

evripidis, 2022-07-27:
@knro Unfortunately ekosdebugger is not running properly, so if you have any hint how I could help other way please advice.

knro, 2022-07-27:
Run the driver via gdb in the terminal:
`gdb -ex "set follow-fork-mode child" -ex run --args indiserver -v indi_toupcam_ccd indi_simulator_telescope`

In Ekos, create a new profile, turn off **Auto Connect** and add Telescope Simulator + Toupcam. Set mode to **remote** and leave it to localhost. In Logs, make sure to select Verbose Logging to File, check INDI + CCD. 

Click **Start** then in INDI Control Panel --> Toupcam --> Options, make sure "Driver Debug" is checked. Then go back to Main Control Panel and press "Connect". Check the terminal to see if the driver crashed, if it did, type **backtrace** and paste the trace here. Also attach the KStars log file.

evripidis, 2022-07-27:
After running in terminal:

```
lubuntu@lubuntu-ami:~$ gdb -ex "set follow-fork-mode child" -ex run --args indiserver -v indi_toupcam_ccd indi_simulator_telescope
GNU gdb (Ubuntu 12.0.90-0ubuntu1) 12.0.90
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from indiserver...
(No debugging symbols found in indiserver)
Starting program: /usr/bin/indiserver -v indi_toupcam_ccd indi_simulator_telescope
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2022-07-27T17:04:44: startup: /usr/bin/indiserver -v indi_toupcam_ccd indi_simulator_telescope[Attaching after Thread 0x7ffff7a34740 (LWP 1954) fork to child process 1957]
[New inferior 2 (process 1957)]
[Detaching after fork from parent process 1954]
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1957 rfd=6 wfd=6 efd=7
2022-07-27T17:04:44: Driver indi_simulator_telescope: pid=1958 rfd=8 wfd=8 efd=9
2022-07-27T17:04:44: listening to port 7624 on fd 5
2022-07-27T17:04:44: Local server: listening on local domain at: @/tmp/indiserver
[Inferior 1 (process 1954) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 1957 is executing new program: /usr/bin/indi_toupcam_ccd
2022-07-27T17:04:44: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2022-07-27T17:04:44: Driver indi_simulator_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD                                                                
2022-07-27T17:04:44: Driver indi_simulator_telescope: snooping on GPS Simulator.TIME_UTC                                                                        
2022-07-27T17:04:44: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK                                                                      
2022-07-27T17:04:44: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_SHUTTER
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #0
[Inferior 2 (process 1957) exited with code 0177]
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1959 rfd=7 wfd=7 efd=11
(gdb) 2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #1
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1960 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: stderr EOF
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #2
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1961 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #3
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1962 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: stderr EOF
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #4
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1963 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #5
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1964 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: stderr EOF
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:44: Driver indi_toupcam_ccd: restart #6
2022-07-27T17:04:44: Driver indi_toupcam_ccd: pid=1965 rfd=7 wfd=7 efd=11
2022-07-27T17:04:44: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEP8fitsfilePNS_7CCDChipE
2022-07-27T17:04:44: Driver indi_toupcam_ccd: stderr EOF
2022-07-27T17:04:44: Driver indi_toupcam_ccd: read: Connection reset by peer
[... cut ...]
2022-07-27T17:04:45: Driver indi_toupcam_ccd: stderr EOF
2022-07-27T17:04:45: Driver indi_toupcam_ccd: read: Connection reset by peer
2022-07-27T17:04:45: Driver indi_toupcam_ccd: Terminated after #10 restarts.
2022-07-27T17:05:14: Client 6: new arrival from local pid 1819 (user: 1000:1000) - welcome!
```
 
And in INDI panel is visible only the simulator. After a few moments the popup "no camera detected" pops up. The camera is connected properly and if I run it through the regular way, the error comes up.

knro, 2022-07-27:
This indicate libindi was not properly installed first. Make sure to compile and install latest libindi first before compiling and installing this driver.

evripidis, 2022-07-27:
After failing to install Kstars/INDI though spa (we've talked about this in the indilib.org) on a fresh formatted unit,  I've followed the instruction here https://github.com/indilib/indi to build it manually. Please note that the 2 days ago the "Install Pre-requisites" part has issues, I can recall for sure an issue with the "libdc1394-22-dev".

So, is it there an "easy way" to remove all of them without formatting the system and start from scratch? Or just to remove and install any package required (example libindi) ?

knro, 2022-07-27:
Stable version would be released on the 29th, stay tuned!

evripidis, 2022-07-29:
> Stable version would be released on the 29th, stay tuned!

@knro Just installed the stable and I got the exact same result, is not loading up and via the terminal command too.

How can I be sure to completely remove INDI to avoid issues?

knro, 2022-07-29:
indi-toupcam not built yet, they would be in 1-2 hours.

evripidis, 2022-07-29:
I've seen that intermittent 2 times. Trying to "replicate" it and catch it with Ekos debugger. I'm just mention it now, probably  will be helpful. 
[image]

knro, 2022-07-29:
This isn't a crash, it's an error so you can't catch that with Ekos Debugger (That's just for crashes).

evripidis, 2022-07-29:
Oh, ok, copy that. Thankfully I've already turned on the logs.

Patrick-81, 2022-07-30:
Yaaaay the toupcam driver works fine. Unfortunately the asi driver still crash
[log_08-41-26.txt](https://github.com/indilib/indi/files/9225506/log_08-41-26.txt)
.

evripidis, 2022-07-30:
No error till now so I could say also issue resolved.

closed 2022-07-30 by evripidis

Klyress, 2022-08-26:
I got the exact same issue on the latest version. Could be a regression?