indilib/indi #1719: INDI produce invalid FITS files

pchev, 2022-08-14:
**Describe the bug**
FITS files produced by INDI are invalid because there is extra data at the end of the file.  
Many software only read the first HDU are not affected, but other reject the file.

For example fpack reject the file:

     Error processing file: IMAGE_007.fits
     in HDU number 1
     FITSIO status = 108: error reading from FITS file
     Error reading data buffer from file:
     IMAGE_007.fits
     Failed to move to HDU number 2 (ffmahd).
     Input file is unchanged.

fitsverify report the error:

             fitsverify 4.20 (CFITSIO V4.000)              
              --------------------------------              
     File: IMAGE_007.fits
     1 Header-Data Units in this file.
     =================== HDU 1: Primary Array ===================
      44 header keywords
      16-bit integer pixels,  2 axes (1280 x 1024), 
     < End-of-File >
     *** Error:   File has extra byte(s) after last HDU at byte 2629440.
     ++++++++++++++++++++++ Error Summary  ++++++++++++++++++++++
      HDU#  Name (version)       Type             Warnings  Errors
      1                          Primary Array    0         0     
      End-of-file                                 0         1     
     **** Verification found 0 warning(s) and 1 error(s). ****

**To Reproduce**
- Run a server with ccd and telescope simulator
- set upload=local
- take an exposure
- run fitsverify with the produced file

The issue is present with last master and has been introduced by e74f9e1ae5c701bc645d81c85e94fa9da9c3c357 
It work normally when using the previous commit.

I cannot propose a PR because I don't know why this change was done.

knro, 2022-08-14:
I see the issue. This was done in order to preallocate memory for the FITS file in memory with the new fast BLOB feature in INDI. Perhaps we can allocate less and let cfitsio just realloc as necessary then? What do you think @pludov ?

pludov, 2022-08-14:
I think in openFITSFile, m_fitsmemorysize should be reverted to hardcoded 2880, and the size argument only be used for IDSharedBlobAlloc

closed 2022-08-28 by knro