indilib/indi #1723: Previously saved values in the INDI Control Panel are no longer available!

AlbertFetscher, 2022-08-25:
Raspberry Pi 4
Astroberry (updated 25.08.2022)
Connected devices:
USB3.0 SanDisk Ultra 32GB memory stick
USB2.0 iOptron GEM28
USB2.0 iOptron iGuider
Connected via VNC client over WLAN

Test steps:

    Start KStars
    KStars -> button "Toggle EKOS"
    EKOS -> button "Start"
    INDI Control Panel, tab "V4L2 CCD":
    a) Note: Options previous set are not available!
    b) Set some options again and saved it:
    b1) "Capture Options": set "Capture Size" to 1280x960
    b2) "Image Info": set "Max. Width", "Max. Height", "Pixel size (um)", "Pixel size X", "Pixel soze Y" to the correct values: 1280, 960, 3.75, 3.75, 3.75 and then click button "Set" (of course)
    b3) "Image Settings": "Frame" indicates red, therefore need to click button "Set"
    b4) "Options": "Configuration" click button "Save" to save configuratoin (of course)
    EKOS -> "Disconnect"
    Close window "EKOS"
    Close window "KStars"
    Reboot Astroberry... and wait until VNC client is ready to reconnect
    Repeat test steps 1, 2 and 3
    INDI Control Panel, tab "V4L2 CCD":
    a) Check if all previous set options (test step 4) are available

Expected result:
Previous set options are available.

Result:
Options previous saved are not available!

Issue:
Test step 10a, in a previous session I set the correct options, saved it but now these values are not available. Why?

(If you need screenshots, don't hesitate to ask.)

knro, 2022-08-28:
driver XML file contains these values?

AlbertFetscher, 2022-08-28:
No, the corresponding INDI driver XML file does not contain the above mentioned key/value pairs related to the INDI control panel tab "V4L2 CCD", and its sub-tabs: 
a) "Capture Options" 
b) "Image Info".

Therefore I can not save these settings, because they are not written to the XML file.

I tried to manually insert der key/value pairs for "Image Info" and INDI/EKOS reads them successfully. But if we click on "save" or any "set" button these key/value pairs are lost and not stored back to the XML file. 

Background: 
I would like to modifiy the pixel size values and pixel width/height for my iOptron iGuider camera (in a permanent manner).

knro, 2022-08-29:
Not all INDI properties are saved to the XML file, only some select properties.

AlbertFetscher, 2022-08-30:
Thanks for your quick reply.
Yes, in this case not all INDI properties are saved in the XML file, only some selected properties. But why is that with the "iOptron iGuider" camera, if I may ask?

E.g. I would like to fix the incorrect default properties, such as:
   CCD_MAX_X = 1280 (instead of 640)
   CCD_MAX_Y = 960
   CCD_PIXEL_SIZE = 3.75 (instead of 5.6)
   CCD_PIXEL_SIZE_X = 3.75 (instead of 5.6)
   CCD_PIXEL_SIZE_Y = 3.75 (instead of 5.6)
and
   Capture Size = 1280x960 (instead of 640x960)

Is this current (strange and currently not understandable for me) behavior due to INDI or EKOS?

knro, 2022-08-30:
Capture size is actually saved IIRC. For pixel size, what camera is this exactly? It can be added to the list to automatically set the correct size, and it should alert you about this.

AlbertFetscher, 2022-08-30:
The camera used is an
  iOptron iGuider, 
  focal length 120mm, 
  aperture 30mm,
  maximum amount of pixels is 1280 x 960 pixels (width x height)
  pixel size is 3.75 µm

Thank you very much!

knro, 2022-11-22:
Ok iOptron pixel sizes is already in the driver... is it getting reset every time you connect?

luzpaz, 2024-02-04:
Is this still relevant ?

closed 2024-03-17 by knro