indilib/indi #1725: Multiple Baseclient instances crash - segmentationfault

gehelem, 2022-08-31:
I'm playing with BaseClient with multiple instances, each instance is responsible for a specific task (focus, guide, ...)
Recently (since 1.9.7 release ?? not sure) my application crashes (segmentationfault) after a while when multiple blobs are catched by each baseclient instance...

**To Reproduce**
I've made a simple test app, from generic-client example.
The only difference is that it instanciates **3** (identical) clients...

1. Build attached test app
2. Run indiserver with CCD Simulator
3. Run attached test app
4. take exposure via indi control panel
5. test app crashes when blob arrives

 - OS: linux kubuntu 20.04
 - indilib 1.9.7

I'm not sure if my way to play with all that is ok, but any help to see what should be done would be appreciated.
I've been playing with that kind of process for months without problems, it seems to be related to latest release 1.9.7.
I've tried with (very old) 1.8.2 it's ok.
I'll try to retrieve previous version 1.9.6 to be sure it related to 1.9.7
I've seen many things happened to blob managment lately, it could be related 

thanks for your help
G.

[generic-client.zip](https://github.com/indilib/indi/files/9463285/generic-client.zip)

gehelem, 2022-08-31:
Sorry, forgot to say hello...
"Hello !"

gehelem, 2022-08-31:
It looks like this one afterwards :
[https://github.com/indilib/indi/issues/1528](url)
crash log is the same :
[image]

knro, 2022-09-01:
I can confirm the crash. @pludov it crashes when we have clients in multiple threads trying to access the data. I believe this is due to IDSharedBlobFree(tmp) at line 756 of basedevice.cpp

pludov, 2022-09-01:
There is a bug in the way shared blob are handled into client, leading to one thread freeing all blobs in that case. (and leak).
I push  a fix here: https://github.com/indilib/indi/pull/1726

By the way, thank @gehelem  for the quality of your report ! Your simple client made fixing this a breeze !

gehelem, 2022-09-01:
Thank you very much @pludov & @knro  !!
i'll to test your fix

gehelem, 2022-09-08:
@pludov  i've tested your modification with my own project, it's ok : thank you very much !

what's next ? should i close this issue, or will you do it later ?

pludov, 2022-09-08:
Hi ! Since the change has already been merged, it seems fine to close this !