indilib/indi #1727: indi_asi_single_ccd crashed

juandanielmorcilloregueiro, 2022-09-02:
Hi,

after update to KStars 3.6.0 my ZWO camera drivers started to crash at start. Initially I though it was an [ identified issue with libASICamera2 file](https://indilib.org/forum/ekos/12051-kstars-3-6-0-release.html?start=24) but it seems only affects to 64bits system, and I´m using a 32 bits system.

It worked fine with 3.5.9 version of KStars.

**SO:** 
Rasbperry Pi 3 with Raspbian Buster 32 bits
Linux astroberry 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

**Logs:**
[2022-09-02T16:12:34.938 CEST DEBG ][ org.kde.kstars.indi] - Adding managed driver "ZWO Camera"
[2022-09-02T16:12:34.939 CEST DEBG ][ org.kde.kstars.indi] - Starting INDI Driver "indi_asi_single_ccd"
[2022-09-02T16:12:34.944 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "FIFO: start indi_asi_single_ccd -n \"ZWO Camera 2\""
[2022-09-02T16:12:34.945 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "With name: ZWO Camera 2"
[2022-09-02T16:12:34.945 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "FIFO: Starting driver indi_asi_single_ccd"
[2022-09-02T16:12:34.946 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "2022-09-02T14:12:34: Driver indi_asi_single_ccd: pid=2076 rfd=10 wfd=15 efd=16"
[2022-09-02T16:12:34.946 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: ""
[2022-09-02T16:12:35.059 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "2022-09-02T14:12:35: Driver indi_asi_single_ccd: indi_asi_single_ccd: symbol lookup error: indi_asi_single_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEPNS_7CCDChipE"
[2022-09-02T16:12:35.059 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: ""
[2022-09-02T16:12:35.062 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "2022-09-02T14:12:35: Driver indi_asi_single_ccd: stderr EOF"
[2022-09-02T16:12:35.062 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: ""
[2022-09-02T16:12:35.062 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "2022-09-02T14:12:35: Driver indi_asi_single_ccd: Terminated after #0 restarts."
[2022-09-02T16:12:35.063 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: "Child process 2076 died"
[2022-09-02T16:12:35.063 CEST DEBG ][ org.kde.kstars.indi] - INDI Server: ""
[2022-09-02T16:12:35.063 CEST CRIT ][ org.kde.kstars.indi] - INDI driver "indi_asi_single_ccd" crashed!

I found a [thread about the same issue](https://indilib.org/forum/ccds-dslrs/12134-indi-asi-ccd-crashes-need-help.html#86007) but there is not solution.

Thanks.

knro, 2022-09-03:
The log shows a link error, so it was not built properly against libindi. What's the source for this? self-compiled or PPA or what?

juandanielmorcilloregueiro, 2022-09-03:
Hi,

I did the update runngin "sudo apt-get upgrade" on my Astroberry distro. I suppose it uses a PPA, I didn´t compile manually.

knro, 2022-09-03:
Ok please contact Astroberry then about this issue, it's not an INDI issue.

closed 2022-09-03 by knro