indilib/indi #1729: Indi server does not automatically save photos

rigelinorion, 2022-09-12:
Hello. Indi server does not automatically save photos either to a local folder (only the FITS preview window is shown) or to a remote one ("Unable to save image file (/home/pi/Pictures\Light/Light_007.fits). Permission denied").

TallFurryMan, 2022-09-12:
Most possibly one element of the path does not exist or cannot be created. What is that backslash in the middle of the string?

rigelinorion, 2022-09-12:
Exactly. This backslash. I don't know where it comes from.

rigelinorion, 2022-09-12:
I wonder which part of the code is responsible for building the path?

TallFurryMan, 2022-09-14:
The path is built from the upload settings property in the INDI panel, in this case the upload dir seems to be `/home/pi`, and the prefix appears to be `Pictures\Light/Light_XXX`. The `XXX` is replaced with the number of similar files already in the upload dir path, and appended with the image format. Do you see where those values are coming from? Is the client KStars or another application?

rigelinorion, 2022-09-15:
I am using the KStars client. I looked into the code, but I can't find where the prefix "\ Light" is inserted into the prefix (depending on the frame type). It definitely builds the path in indiccd.uploadFile. At least you were right, it was a permissions problem. After creating a new folder, EKOS creates a folder with the backslash and saves the photos there.

TallFurryMan, 2022-09-15:
KStars client builds the file path from different values. You may have a weird value somewhere there. So you needed to create the "Light" folder? I remember the KStars client requires "Pictures" to exist and will only create subfolders. But here it is the local rpi indiserver that saves the file, and there is no subfolder creation procedure apparently.

TallFurryMan, 2022-09-15:
Feel free to submit a PR if you think the code lacks this ability!

rigelinorion, 2022-09-18:
There is nowhere in the indiserver code to add a subfolder to the path. It looks like EKOS sets this subfolder by adding a backslash. Every time EKOS is loaded, this unfortunate subfolder with the \ sign is added to the path.
  It looks like I will have to edit the saving path of the files each time I boot.

knro, 2022-09-19:
This is not an INDI issue, but an Ekos issue. Please file with bugs.kde.org

closed 2022-09-19 by knro