indilib/indi #1736: indiccd.ccp: The expression to initialize isON in CCD::addCaptureFormat function is incorrect.

jctk, 2022-09-27:
**Describe the bug**
The following line in function `CCD::addCaptureFormat` is incorrect.
```C++
    const bool isOn = (format.label == m_ConfigCaptureFormatLabel) || (m_ConfigCaptureFormatLabel.empty() && format.isDefault);
```
The `m_ConfigCaptureFormatLabel` is assigned the name attribute of the onSwitch that is set to ON in the "CCD_CAPTURE_FORMAT" of the config file.
For example, in the case of ZWO ASI CCD, it is a string such as "ASI_IMG_RAW16".
On the other hand, `format.label` is assigned to the label part of ISwitch, i.e., a string such as "Raw 16 bit".
Therefore, if `m_ConfigCptureFomatLabel` is assigned a value from a config file, isOn is always false.

**Expected behavior**
Correct the expression as follows.
```C++
    const bool isOn = (format.name == m_ConfigCaptureFormatLabel) || (m_ConfigCaptureFormatLabel.empty() && format.isDefault);
```

**Screenshots**
[image]

knro, 2022-09-27:
Thanks! I pushed a fix, it was a bit more involved but you were absolutely right on the name vs. label issue.

closed 2022-09-27 by knro

jctk, 2022-09-28:
Isn't it safe that configName doesn't terminate with `\0' if the length of dvice name is 64 characters, which is the same as the size of configName?

indiccd.cpp line 357-358
```C++
    char configName[64] = {0};
    if (IUGetConfigOnSwitchName(getDeviceName(), "CCD_CAPTURE_FORMAT", configName, MAXINDINAME) == 0)
```

I think it is better to reserve the size of configName for '\0' and assign '\0' to the position of the last character in advance as follows.  It is even better if the IUGetConfigOnSwitchName function assigns '\0' to the last character position.

```C++
    char configName[MAXINDINAME+1] = {0};
    configName[MAXINDINAME] = '\0';
```

commit d3e66830 (2022-09-27): Fix using label when name should be used instead since it is the only attribute saved in the configuration file. Fixes #1736