indilib/indi #1749: SkySafari Client/Server and DEC between -00:00:01 and -00:59:59

astrogene1000, 2022-10-28:
As identified here:
https://indilib.org/forum/mounts/12517-declination-inverted-in-skysafari-driver-client.html

In Ekos, I just set Telescope = Telescope Simulator and Aux = skysafari.

Then pointed an -ancient- version of Skysafari to IP:9624 and connected.

Tried a GoTo star 'Heze' @ -00:42:41.4

Scope Simulator was pointed to +00:42:41

The log's show:
[2022-10-27T12:53:46.360 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] Current RA: 13:35:51 Current DEC: 0:42:41 "
[2022-10-27T12:53:46.608 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] mountToApparentRaDec 8.502344, 0.711389 to ha 8.502344, ra -156.038491, 0.711389 "
[2022-10-27T12:53:46.609 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] HaAxis: 8.502344, ra 13.597434 "
[2022-10-27T12:53:46.610 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] DecAxis: 0.711389, dec 0.711389 "
[2022-10-27T12:53:46.611 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] Current RA: 13:35:51 Current DEC: 0:42:41 "
->>>>[2022-10-27T12:53:46.751 EDT DEBG ][ org.kde.kstars.indi] - SkySafari : "[DEBUG] CMD <Sd-00*42:41> "
[2022-10-27T12:53:46.752 EDT DEBG ][ org.kde.kstars.indi] - SkySafari : "[DEBUG] RES <1> "
[2022-10-27T12:53:46.752 EDT DEBG ][ org.kde.kstars.indi] - SkySafari : "[DEBUG] CMD <> "
[2022-10-27T12:53:46.859 EDT DEBG ][ org.kde.kstars.indi] - Telescope Simulator : "[SCOPE] mountToApparentRaDec 8.503391, 0.711389 to ha 8.503391, ra -156.038491, 0.711389 "

So Skysafari is sending correct info

The skysafari server code shows:
else if (cmd.compare(0, 2, "Sd") == 0)
{
int dd = 0, mm = 0, ss = 0;
if (sscanf(cmd.c_str(), "Sd%d*%d:%d", &dd, &mm, &ss) == 3)
{
DE = abs(dd) + mm / 60.0 + ss / 3600.0;
->>> if (std::signbit(dd))
DE *= -1;
}

They are trying to get the sign of the 'dd' but sscanf cannot return it.

A fix would be to replace
if (std::signbit(dd))
With
if(strncmp(cmd.c_str(),"Sd-",3) == 0)

Gene

knro, 2022-10-28:
Where is this code? which file?

astrogene1000, 2022-10-29:
Sorry, false alarm, this -was fixed- in v1.9.5

closed 2022-10-29 by astrogene1000