indilib/indi #1755: Efficient camera streaming

knro, 2022-11-09:
Currently INDI camera drivers rely on StreamManager to send RAW and JPG images back to clients. This can be extremely inefficient even when the client is on the same host. Which options can we support that would make video streaming (and perhaps recording) more efficient?

+ MJPEG server?
+ WebRTC end point?
+ RTSP?

@pawel-soja @pchev @pludov What are you thoughts on this matter long-term?

pawel-soja, 2022-11-10:
I also propose to consider data transfer based on [MessagePack](https://msgpack.org/).
Benefits:
- all xml communication can be moved
- easy encode / decode to Json
- support for typed data in JavaScript
- 1:1 binary data can be transferred
- using the WebSocket protocol, easy creation of direct web clients

pludov, 2022-11-11:
Hi !

I asssume you are more talking about view/preview than data acquisition -
for which lossy compression is probably never wished. For lossless
acquisition, you have to transfert all data anyway -  compression is a
slowdown - but fortunately sharedbuffer already do this with almost max
perf (still with some win possible in the drivers), so for me this is a
closed case.

For lossy/Preview, we probably want an efficient lossy compression plus
visualisation settings like min max, gamma, compression quality,
scaling,..  .

My advice would be to build a dedicated indi client that exposes stream
over http (mjpeg or h26x...) on top of indiserver, but not part of it.
You'd have one endpoint (url) per indi blob + endpoint for tweaking their
rendering settings. Above that you'd build preview widget : one for
web/JavaScript context, one for qt, one for flutter, ...

The good reasons for keeping it put of indiserver ( and drivers) i can
think of :
  - robustness: video encoding is sometime prone to crash  ( esp when hw
acceleration is used, or clients behavior may be hard to control, like web
browser which can open multiple requests ....)
  - independency: could be built on any tech, like python for taking
advantage of image processing lib for example
  - can work concurrently with a full res / lossless capture (frame would
be shared between the preview and the capture client, the frame dropped by
the preview would not be lost by the client since arbitration occurs in
indiserver)
  - could be extended for live view of other data ( like live stacking)

I've implemented something similar for mobindi - Fitsviewer . It is a pure
c++ indi client for displaying captures on a web view. it was initially
more intended for efficient display of still images in browser over wifi,
but it can now also display streams with reasonnable perfs on a rpi+ phone
setup ( incl histogram, debayer, ability to zoom efficiently, adaptation of
encoding quality to resolution...)

knro, 2022-11-11:
@pawel-soja This is very interesting but perhaps a bit out of the scope of this discussion and would require massive efforts.

@pludov I agree 100%! This is the approach I was thinking of, but what would be a nice solution to have? A client that receives images, and create an [mpeg-streamer](https://github.com/jacksonliam/mjpg-streamer) server?

closed 2022-11-22 by knro