indilib/indi #1765: Throttle FPS update?

knro, 2022-11-15:
Is this possible to throttle FPS updates to like once per second? It appears it is sending it too fast now and can cause unnecessary traffic at the client side. For some cameras where FPS is like 200 fps, we don't need to update the FPS value at 200 Hz!

https://github.com/indilib/indi/blob/e425d43bac5f716af11ecab10b2c2588d435f036/libs/stream/streammanager.cpp#L294

pawel-soja, 2022-11-15:
Have you found a real problem? I can take a closer look, but just had a quick look at the code:

From what I remember, FPS is calculated in 2 ways. Every 1 second and every 100ms.
https://github.com/indilib/indi/blob/e425d43bac5f716af11ecab10b2c2588d435f036/libs/stream/streammanager.cpp#L47-L48

knro, 2022-11-15:
Ah okay, so it can be changed from here. For embedded devices, maybe we can set both to 1000. I'll add an ifdef

closed 2022-11-15 by knro

knro, 2022-11-15:
I'm experimenting now with a few values to arrive at a good compromise.

pawel-soja, 2022-11-15:
For a reminder of old times...
I struggled with the RPi4 and the biggest problem was copying the movie frame multiple times from memory to memory, especially with ASI drivers (the next step was to be more optimization of StreamManager). I was unable to establish cooperation with the ZWO.
Maintaining [AsiCamera](https://github.com/pawel-soja/AsiCamera) is troublesome because I don't have many cameras to test. My current solution is a large USB 3.0 hub and a long cable to the computer.

pawel-soja, 2022-11-15:
in summary, the RPi has too weak a RAM clock to cope with fast cameras with reasonable resolution.