indilib/indi #1778: INDI Client crash on MacOS

knro, 2022-12-02:
All users who tested MacOS KStars build reported that KStars crashes when connecting to a remote INDI server. I can confirm this issue and got this backtrace:

```
2022-12-03 00:05:58.440291+0300 kstars[39302:2910024] [org.kde.kstars.indi] NightCrawler :  "[INFO] Focuser Type 2.5 NC "
2022-12-03 00:05:58.440724+0300 kstars[39302:2910024] [org.kde.kstars.indi] NightCrawler :  "[INFO] NightCrawler is online. "
Process 39302 stopped
* thread #32, stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x00007ff80acd7572 libsystem_platform.dylib`_platform_strlen + 18
libsystem_platform.dylib`:
->  0x7ff80acd7572 <+18>: pcmpeqb (%rdi), %xmm0
    0x7ff80acd7576 <+22>: pmovmskb %xmm0, %esi
    0x7ff80acd757a <+26>: andq   $0xf, %rcx
    0x7ff80acd757e <+30>: orq    $-0x1, %rax
Target 0: (kstars) stopped.
(lldb) bt
* thread #32, stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x00007ff80acd7572 libsystem_platform.dylib`_platform_strlen + 18
    frame #1: 0x000000010e4e1ecf libindiclient.1.9.9.dylib`INDI::LilXmlValue::toDouble(INDI::safe_ptr<bool>) const [inlined] std::__1::char_traits<char>::length(__s=0x0000000000000000) at __string:342:53 [opt]
    frame #2: 0x000000010e4e1ec7 libindiclient.1.9.9.dylib`INDI::LilXmlValue::toDouble(INDI::safe_ptr<bool>) const [inlined] std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(this=Summary Unavailable, __s=0x0000000000000000) at string:842:19 [opt]
    frame #3: 0x000000010e4e1ec7 libindiclient.1.9.9.dylib`INDI::LilXmlValue::toDouble(INDI::safe_ptr<bool>) const [inlined] std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(this=Summary Unavailable, __s=0x0000000000000000) at string:840:88 [opt]
    frame #4: 0x000000010e4e1ec7 libindiclient.1.9.9.dylib`INDI::LilXmlValue::toDouble(this=<unavailable>, ok=<unavailable>) const at indililxml.h:252:28 [opt]
    frame #5: 0x000000010e4df6b5 libindiclient.1.9.9.dylib`INDI::BaseDevice::buildProp(INDI::LilXmlElement const&, char*, bool) [inlined] INDI::LilXmlValue::operator double(this=0x0000000315a22300) const at indililxml.h:88:44 [opt]
    frame #6: 0x000000010e4df6a2 libindiclient.1.9.9.dylib`INDI::BaseDevice::buildProp(this=<unavailable>, root=<unavailable>, errmsg=<unavailable>, isDynamic=false) at basedevice.cpp:466:29 [opt]
    frame #7: 0x000000010e4e5d29 libindiclient.1.9.9.dylib`INDI::WatchDeviceProperty::processXml(this=<unavailable>, root=0x0000000315a225e8, errmsg=<unavailable>, constructor=<unavailable>)> const&) at watchdeviceproperty.cpp:141:34 [opt]
    frame #8: 0x000000010e4f97ee libindiclient.1.9.9.dylib`INDI::AbstractBaseClientPrivate::dispatchCommand(this=0x000000011a014800, root=0x0000000315a225e8, errmsg=<unavailable>) at abstractbaseclient.cpp:117:24 [opt]
    frame #9: 0x000000010e4cefaa libindiclient.1.9.9.dylib`std::__1::__function::__func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator()(char const*&&, unsigned long&&) at baseclient.cpp:252:28 [opt]
    frame #10: 0x000000010e4cef1c libindiclient.1.9.9.dylib`std::__1::__function::__func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator()(char const*&&, unsigned long&&) [inlined] decltype(__f=<unavailable>, __args=<unavailable>, __args=<unavailable>)::$_0&>(fp)(static_cast<char const*>(fp0), static_cast<unsigned long>(fp0))) std::__1::__invoke<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0&, char const*, unsigned long>(INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0&, char const*&&, unsigned long&&) at type_traits:3918:1 [opt]
    frame #11: 0x000000010e4cef12 libindiclient.1.9.9.dylib`std::__1::__function::__func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator()(char const*&&, unsigned long&&) [inlined] void std::__1::__invoke_void_return_wrapper<void, true>::__call<INDI::BaseClientPrivate::BaseClientPrivate(__args=<unavailable>, __args=<unavailable>, __args=<unavailable>)::$_0&, char const*, unsigned long>(INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0&, char const*&&, unsigned long&&) at invoke.h:61:9 [opt]
    frame #12: 0x000000010e4cef12 libindiclient.1.9.9.dylib`std::__1::__function::__func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator()(char const*&&, unsigned long&&) [inlined] std::__1::__function::__alloc_func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator(this=<unavailable>, __arg=<unavailable>, __arg=<unavailable>)(char const*&&, unsigned long&&) at function.h:178:16 [opt]
    frame #13: 0x000000010e4cef12 libindiclient.1.9.9.dylib`std::__1::__function::__func<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0, std::__1::allocator<INDI::BaseClientPrivate::BaseClientPrivate(INDI::BaseClient*)::$_0>, void (char const*, unsigned long)>::operator(this=<unavailable>, __arg=<unavailable>, __arg=<unavailable>)(char const*&&, unsigned long&&) at function.h:352:12 [opt]
    frame #14: 0x000000010e4fdfd5 libindiclient.1.9.9.dylib`TcpSocket::readyRead() [inlined] std::__1::__function::__value_func<void (char const*, unsigned long)>::operator(this=<unavailable>, __args=0x0000000315a22eb8, __args=0x0000000315a22eb0)(char const*&&, unsigned long&&) const at function.h:505:16 [opt]
    frame #15: 0x000000010e4fdfc1 libindiclient.1.9.9.dylib`TcpSocket::readyRead() [inlined] std::__1::function<void (char const*, unsigned long)>::operator(this=<unavailable>, __arg="<defNumberVector device=\"NightCrawler\" name=\"FOCUS_STEP_DELAY\" label=\"Step Rate\" group=\"Settings\" state=\"Idle\" perm=\"rw\" timeout=\"0\" timestamp=\"2022-12-02T21:05:58\">\n    <defNumber name=\"FOCUS_STEP\" label=\"Value\" format=\"%.f\" min=\"7\" max=\"100\" step=\"1\">\n7\n    </defNumber>\n</defNumberVector>\n<defLightVector device=\"NightCrawler\" name=\"LIMIT_SWITCHES\" label=\"Limit Switch\" group=\"Settings\" state=\"Idle\" timestamp=\"2022-12-02T21:05:58\">\n    <defLight name=\"ROTATION_SWITCH\" label=\"Rotation Home\">\nOk\n    </defLight>\n    <defLight name=\"OUT_SWITCH\" label=\"Focus Out Limit\">\nOk\n    </defLight>\n    <defLight name=\"IN_SWITCH\" label=\"Focus In Limit\">\nOk\n    </defLight>\n</defLightVector>\nET_3\" label=\"Preset 3\" format=\"%.f\" min=\"0\" max=\"100000\" step=\"1000\">\n0\n    </defNumber>\n</defNumberVector>\n<defSwitchVector device=\"NightCrawler\" name=\"Goto\" label=\"Goto\" group=\"Presets\" state=\"Idle\" perm=\"rw\" rule=\"OneOfMany\" timeout=\"0\" timestamp=\"2022-12-02T21:05:58\">\n    <defSwitch name=\"Preset 1\" label=\"Preset 1\">\nOff\n    </defSwitch"..., __arg=683)(char const*, unsigned long) const at function.h:1182:12 [opt]
    frame #16: 0x000000010e4fdfc1 libindiclient.1.9.9.dylib`TcpSocket::readyRead() [inlined] TcpSocket::emitData(this=<unavailable>, data="<defNumberVector device=\"NightCrawler\" name=\"FOCUS_STEP_DELAY\" label=\"Step Rate\" group=\"Settings\" state=\"Idle\" perm=\"rw\" timeout=\"0\" timestamp=\"2022-12-02T21:05:58\">\n    <defNumber name=\"FOCUS_STEP\" label=\"Value\" format=\"%.f\" min=\"7\" max=\"100\" step=\"1\">\n7\n    </defNumber>\n</defNumberVector>\n<defLightVector device=\"NightCrawler\" name=\"LIMIT_SWITCHES\" label=\"Limit Switch\" group=\"Settings\" state=\"Idle\" timestamp=\"2022-12-02T21:05:58\">\n    <defLight name=\"ROTATION_SWITCH\" label=\"Rotation Home\">\nOk\n    </defLight>\n    <defLight name=\"OUT_SWITCH\" label=\"Focus Out Limit\">\nOk\n    </defLight>\n    <defLight name=\"IN_SWITCH\" label=\"Focus In Limit\">\nOk\n    </defLight>\n</defLightVector>\nET_3\" label=\"Preset 3\" format=\"%.f\" min=\"0\" max=\"100000\" step=\"1000\">\n0\n    </defNumber>\n</defNumberVector>\n<defSwitchVector device=\"NightCrawler\" name=\"Goto\" label=\"Got [... cut ...]
    frame #22: 0x000000010e4fe2b4 libindiclient.1.9.9.dylib`void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, TcpSocketPrivate::connectToHost(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short)::$_0, std::__1::thread> >(__vp=0x0000600001714900) at thread:298:5 [opt]
    frame #23: 0x00007ff80acaf259 libsystem_pthread.dylib`_pthread_start + 125
    frame #24: 0x00007ff80acaac7b libsystem_pthread.dylib`thread_start + 15
```

Any ideas @pawel-soja ?

pawel-soja, 2022-12-03:
It seems that an exception is not thrown when the string is null. I'm going to make coffee and make corrections soon.

pawel-soja, 2022-12-03:
Fixed: https://github.com/indilib/indi/pull/1779

closed 2022-12-05 by knro